arduino/ArduinoCore-mbed #546: PluggableUSBAudio not working

w9ran, 2022-09-03:
Trying this example:
/*
 * Copyright (c) 2018-2019, Arm Limited and affiliates.
 * SPDX-License-Identifier: Apache-2.0
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include "Arduino.h"

#include <stdint.h>
#include <string.h>
#include "PluggableUSBAudio.h"
#include "USBAudio_Types.h"
#include "EndpointResolver.h"
#include "usb_phy_api.h"

using namespace arduino;

#warning "USBAudio library is EXTREMELY EXPERIMENTAL, expect crashes"

#define SAMPLE_SIZE                 2
#define XFER_FREQUENCY_HZ           1000
#define WRITE_READY_UNBLOCK         (1 << 0)
#define READ_READY_UNBLOCK          (1 << 1)

//#define lock()
//#define unlock()
//#define assert_locked()
//#define assert_unlocked()

class USBAudio::AsyncWrite: public AsyncOp {
public:
    AsyncWrite(USBAudio *audio, uint8_t *buf, uint32_t size):
        audio(audio), tx_buf(buf), tx_size(size), result(false)
    {

    }

    virtual ~AsyncWrite()
    {

    }

    virtual bool process()
    {
        if (audio->_tx_state != Opened) {
            result = false;
            return true;
        }

        uint32_t actual_size = 0;
        audio->write_nb(tx_buf, tx_size, &actual_size);
        tx_size -= actual_size;
        tx_buf += actual_size;
        if (tx_size == 0) {
            result = true;
            return true;
        }

[... 914 lines cut ...]

    // Start the write
    PluggableUSBD().write_start(_episo_in, _tx_packet_buf, send_size);
    _tx_idle = false;
    _tx_frame_fract += _tx_fract_frames_per_xfer;
}

void USBAudio::_send_isr()
{
    assert_locked();

    write_finish(_episo_in);

    _send_isr_next_sync();

    // Signal that there is space for more data
    _write_list.process();
    if (_tx_done) {
        _tx_done.call(Transfer);
    }
}

But the Pico continues to enumerate as a USB Serial device, not as Mbed Audio.
Thanks, 
Bob

facchinm, 2022-09-05:
Hi @w9ran ,
it should enumerate as a composite device with both the Serial (CDC) functionality and the USBAudio one. 
On Windows, a new driver scan should be automatically started thanks to https://github.com/arduino/ArduinoCore-mbed/commit/301d69d045d9f3e636ec4f8a1f8d5b02b2cb0e3f and you should see the new audio interface without any further intervention.
If it's not the case let me know and I'll try to reproduce the issue.

w9ran, 2022-09-06:
Thanks for your reply and suggestion.  I did prove that the Pico is sending audio via USB in Linux, so it's a Windows issue.  

I deleted all Pico containers and attached the Pico afresh so Windows did the usual "setting up new device" routine.  This resulted in three Pico entries in device manager, all identical and with the same "Code 10 - cannot start" error:

[image]

I tried updating the driver but Windows said the best driver was installed (USB Audio) - here is are the details:

[image]

I then tried disabling/re-enabling the device which produced a message "Device is enabled but still not working".

Adding a print statement to the main loop proved that during all the above the COM port remained connected and working.

I hope this helps to identify what's going on as having a composite serial/USBAudio device will be very nice.

Bob

facchinm, 2022-09-06:
This page https://docs.microsoft.com/en-us/answers/questions/549388/windows10-usb-problem-0xc0000182-audio-vcom-compos.html might be explaining the issue but no solution is provided unfortunately :disappointed:

w9ran, 2022-09-07:
Yes, the "updates" suggested there were from 2017.   
I can only add this:    In Linux, the Pico presents itself as an audio device just fine,  i.e. arecord -l:

card 2: Pico [RaspberryPi Pico], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

But when I attach the Pico in /dev I see /dev/ttyACM0 rather than something like /dev/ttyUSB0 that I was expecting.    Apparently this is normal and I don't see how it could be affecting USBAudio on Windows since it sees the COM Port OK, but thought I'd mention it.

Bob

MNS26, 2022-09-08:
@w9ran have you tried using tinyusb? It also has a template for usb descriptors (aka how a OS sees the device)
i personally havent tried/looked into it with audio yet but im using it to make a gamepad with 10 axis 3 hats and a unholy amount of buttons

i used the tinyusb library from adafruit for it

MNS26, 2022-09-08:
also /dev/ttyACM is a serial modem whereas /dev/ttyUSB0 is your generic usb to serial chip (from my understanding)

Gerriko, 2022-09-13:
I'm throwing this in the pot as a guess, as I suspect Windows is seeking out a parameter to inform it that you've just inserted a composite device rather than a single USB device type. I know you can define some USB settings inside code, such as vendor id etc. but no idea what can be done here.

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/register-a-composite-driver

je1rav, 2023-05-26:
The reason why not working on Windows seems to be the lack of IAD descriptor in USBAudio.cpp.
By adding the IAD descriptor, It becomes working on Windows too.

w9ran, 2023-08-14:
je1rav - that's great news!  I just looked at this thread after nearly a year and was pleased to find your comment - but I'm not enough of a USB expert to know exactly how to make the change to add the IAD descriptor.   Could you perhaps post a snippet or the entire modified USBAudio.cpp file?

Thanks!
Bob

je1rav, 2023-08-17:
Dear Bos,

Thnak you for your mail.
I am not familiar with github for the pull request.
But I have uploaded the files to
https://github.com/je1rav/QP7C_RP2040_CAT_Arduino.
In it, I created a amature radio transceiver program using RP2040 and USB
Audio.
I am sorry, but the document is written in Japanese.
Please use the language translator to read it.
I think that the files "PluggableUSBAudio.h" and "USBAudio.cpp" in it could
be used for your perpose.

Best regards,
Hitoshi, JE1RAV

jh4vsm, 2023-08-17:
Dear Bob,

The above address was wrong.
The correct address is https://github.com/je1rav/QP-7C_RP2040_CAT_Arduino

Hitoshi

je1rav, 2023-08-17:
Dear Bob,

The above address was wrong.
The correct address is https://github.com/je1rav/QP-7C_RP2040_CAT_Arduino

Hitoshi