arduino/ArduinoCore-mbed #553: Cannot install the board from IDE anymore

BluuJo, 2022-09-21:
Hello,

Since few days everything messed up while I upgraded to Arduino IDE 2.
I have an Arduino Portenta H7 which worked well since I tried to go to MBED standalone 3.3.0.
When I try to download that from legacy IDE I got an error downloading (see at the bottom of the window):
[image]

And with the new IDE I am just stuck here:
[image]

I am on a macOS Monterey.

I tried so many thing but here are essentials:
- fresh install (after a complete uninstall)
- on a Windows 10 PC but I got a driver issue which allow me to upload but not see the serial monitor...
- Online editor but I cannot upload (only look at serial monitor)

per1234, 2022-09-22:
Hi @BluuJo. I saw you also posted over on the Arduino Forum:

https://forum.arduino.cc/t/arduino-ide-not-working-on-macos-anymore/1034541

We'll see if we can help you out over there. I'll make sure to report back here with our findings.

facchinm, 2022-09-29:
Fixed in the forum thread

closed 2022-09-29 by facchinm