arduino/ArduinoCore-mbed #555: Ethernet.setRetransmissionTimeout, Ethernet.setRetransmissionCount

salasidis, 2022-09-23:
Both of these functions are defined in the Ethernet.h header, but the actual code is nowhere to be found, and I get a linker error

 `
 Ethernet.setRetransmissionTimeout(150);

 Ethernet.setRetransmissionCount(1);
 `

 `
Z:\User\RS\EnviroWatch/EthernetInterface.ino:126: undefined reference to arduino::EthernetClass::setRetransmissionTimeout(unsigned short)
Z:\User\RS\EnviroWatch/EthernetInterface.ino:127: undefined reference to arduino::EthernetClass::setRetransmissionCount(unsigned char)
 `

Is this an oversight or by design?

i-herrera, 2022-09-26:
Ethernet library in the Portenta is based on lwip so not all tunables are available.  It is possible however to limit the number of retries directly from the sketch. For example, if the browser is not connected (eg. the soket is closed) `write` method should return '0'.

salasidis, 2022-10-11:
Would that not require access into the library itself?

My only interacting with the browser is via send commands, and then everything else happens in the async library.