arduino/ArduinoCore-mbed #575: How to code the inner M4 core of the Nicla Vision on the Arudino IDE?

hpssjellis, 2022-11-15:
@sebromero  @facchinm 

As far as I can tell, there is no simple method using either Arduino IDE to program the inner M4 core of the Nicla Vision. On The PortentaH7 you have a choice of which core to program in the board manager. Does anyone have example code that runs on either core of the Nicla Vision?

Image of the PortentaH7 showing how easy it is to program each core

[image]

Image of the Nicla Vision which is not obvious how to program the inner M4 core.

[image]

hpssjellis, 2023-01-04:
Any movement on this issue? I assume using a full C++ setup with makefile this can be done, I just have never seen an example of one.

rneurink, 2023-06-12:
See Examples -> RPC -> RPC_m4
https://github.com/arduino/ArduinoCore-mbed/blob/aef18a14398de1a7871d49a7de63fcb5d3643818/libraries/RPC/examples/RPC_m4/RPC_m4.ino#L1-L110

You can select the different cores from the tools menu.
Make sure to also allocate some flash to the M4 core by selecting one of the options with M4 flash under Tools ->**Flash Split**
This examples needs to be uploaded to both cores to work.
[image]

hpssjellis, 2023-06-12:
thanks @rneurink This can be closed

closed 2023-06-12 by hpssjellis