arduino/ArduinoCore-mbed #578: Nano 33 BLE - No luck mapping Esc, Shift, and Ctrl to buttons

UnavengedAvo, 2022-11-17:
I am currently trying to make a one handed controller for gaming that basically emulates a keyboard instead of a controller, with the Thumbstick emulating WASD and push buttons for other keys. 

I used the base USBHID -> Keyboard example that was included under the Nano 33 BLE, and successfully managed to get most of the keys working in a game after mapping the Keyboard.key_code values based on ASCII codes I found online.

I managed to assign Tab to one of the buttons using the 0x09 hex code, but for the live of me I can't seem to map Esc, Shift, and Ctrl to buttons.

I have tried every iteration of mappings from the USBMouseKeyboard.cpp, including KEY_SHIFT and KEY_CTRL, as well as other sources I had found online, but I just can't get these to work.

Based on what I could find, the following ASCII values should work with these keys, but I just can't get them to work:

Symbol -> Dec -> Hex
Esc -> 27 -> 0x1b
Shift -> 16 -> 0x10
Ctrl -> 17 -> 0x11

Am I possibly doing something wrong here?

facchinm, 2022-12-21:
Hi @UnavengedAvo ,
I pushed https://github.com/arduino/ArduinoCore-mbed/pull/597 to fix the issue about not being able to send keycodes not listed in keymap (https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/USBHID/src/USBKeyboard.cpp#L36).

facchinm, 2022-12-21:
About SHIFT and CTRL, they are modifier keys, so they should be passed as the second parameter of `key_code_raw` function

closed 2023-01-10 by facchinm