arduino/ArduinoCore-mbed #580: Pin definition error in PORTENTA_H7_M4/pins_arduino.h

david-nc, 2022-11-23:
File ArduinoCore-mbed/variants/PORTENTA_H7_M4/pins_arduino.h

The digital pin definitions for the pins D16 through D21 appear to be incorrect, and result in an error when used in code.  These pins correspond to the digital pin designations for the analog pins.

 48 // Digital pins
 49 // -----------
 50 #define D0  (0u)
 51 #define D1  (1u)
 52 #define D2  (2u)
 53 #define D3  (3u)
 54 #define D4  (4u)
 55 #define D5  (5u)
 56 #define D6  (6u)
 57 #define D7  (7u)
 58 #define D8  (8u)
 59 #define D9  (9u)
 60 #define D10 (10u)
 61 #define D11 (11u)
 62 #define D12 (12u)
 63 #define D13 (13u)
 64 #define D14 (14u)
 65 #define D15 (15u)
 66 #define D16 (u16)
 67 #define D17 (u17)
 68 #define D18 (u18)
 69 #define D19 (u19)
 70 #define D20 (u20)
 71 #define D21 (u21)

The corresponding analog pin definitions appear to be correct:

 27 // Analog pins
 28 // -----------
 29 #define PIN_A0 (15u)
 30 #define PIN_A1 (16u)
 31 #define PIN_A2 (17u)
 32 #define PIN_A3 (18u)
 33 #define PIN_A4 (19u)
 34 #define PIN_A5 (20u)
 35 #define PIN_A6 (21u)
 36 #define PIN_A7 (22u)

oqibidipo, 2022-11-23:
Variant NICLA_VISION has the the same issue.

DeveloperMarcial, 2022-11-23:
I assume the **M4** cannot use some digital pins like the **M7**:
From `\PORTENTA_H7_M7\pins_arduino.h`
`#define D15 #error Pin cannot be used as digital pin.`
`#define D16 #error Pin cannot be used as digital pin.`
`#define D17 #error Pin cannot be used as digital pin.`
`#define D18 #error Pin cannot be used as digital pin.`

tjaekel, 2022-11-23:
It is the opposite way: file given with this #error was for H7 (M7). So, H4 (M7) can access but not H7 (M7).
Actually, I am not convinced: the H7 (M7) is the master in this chip, H4 (M4) has limited access to all features.

I can only imagine this #error means:
some pins, e.g. there is PC2 and PC2_C: so, via an analog mux you can connect PC2 with PC2_C (the analog pin).
They might to tell you: you cannot configure PC2_C as a digital pin (it goes via an analog mux to a different pin).

Some analog pins are ADC analog inputs: you can "cross-connect" with the other digital pin, PC2 drives now also PC2_C. but PC2_C is not a digital pin (can never be configured as digital out, just via "internal wiring it gets value from PC2).
Actually, it should be the same for H4 (M4) - not usable as digital pin.
It is just a "bug" in the source code and potentially a "bug" how which core could access analog pins.

hpssjellis, 2022-11-24:
The Portenta HD pins have been in a few issues in the past. Some clarification would be very nice. 

Here are a few links:

[SOLVED] Portenta H7 Lite - Low level pin linkage (A5, D8, D9 to A3) #488

https://github.com/arduino/ArduinoCore-mbed/issues/488

High Density Connector Pin Names #74 

https://github.com/arduino/ArduinoCore-mbed/issues/74

Can the Portenta full pinout .pdf be available as a .csv file? #483

https://github.com/arduino/ArduinoCore-mbed/issues/483

tjaekel, 2022-11-24:
I agree.
I do not go with Arduiono anymore: it is too buggy (and their IDE is too slow).
I am fine to use Portenta H7 with my own BSP and in STM CubeIDE (much better and faster).

TuringTW, 2022-12-16:
Hi @tjaekel 

Could you share some information on how to use STM CubeIDE for Portenta H7?

DeveloperMarcial, 2022-12-16:
@TuringTW Check [this](https://github.com/skjafar/Portenta_Cube_Template).