arduino/ArduinoCore-mbed #581: Using Mbed Socket API

aqw42, 2022-11-23:
Hello, i am trying to port this library : https://github.com/nimbuscontrols/EIPScanner/ 
It's a library to control Ethernet/IP (the industrial protocol) devices.

I used MBedOS Socket API as a Backend instead of POSIX sockets (posix needed lwip to be enable but I can't get it working :( )

I successfully compile (yay) but the portenta crashed when the program tries to open a UDP socket.
My question is : *Can* the Socket API actually be used ?
Did someone achieved to use them without crashing ?

(I'm importing my code into a git, so I can turn it into an Arduino library if I succeed)

Any help would be greatly appreciated, as I have a lot of struggle on this task 

Thanks

manchoz, 2022-11-29:
Hi @aqw42,
do you have any examples or sketches to share?

closed 2022-12-05 by aqw42

hpssjellis, 2022-12-05:
@aqw42 this looks interesting, did you close it because you solved it?

I am trying to send raw/bmp/png images to a webpage using the Portenta and a socket might be the best method.

aqw42, 2022-12-13:
I closed it because i couldn't make it work. I spent weeks on this, the compilation never succeded.
At least now we know why the libmbed.a won't recompile directly from the package in Arduino15, but there is a *shitload* of other double definitions, conflicting libraries, etc

If someone want to try this again, I made some other posts on this subject, on the arduino forum and in this git repo.
Good luck :°