arduino/ArduinoCore-mbed #585: Cant't recompile EDGE_CONTROL mbed core

brinata, 2022-11-29:
I need to add cellular support on MBED core for edge control, the script mbed-os-to-arduino give me back the following erros:

I'm working on windows10

$ ./mbed-os-to-arduino -a EDGE_CONTROL:EDGE_CONTROL

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=0
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/c/Users/Ronni/AppData/local/arduino15/packages/arduino/hardware/mbed_edge/3.4.1

VARIANT=EDGE_CONTROL BOARD=EDGE_CONTROL
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "C:\Users\Ronni\AppData\Local\Temp\mbed-os-program" (program)
[mbed] EDGE_CONTROL now set as default target in program "mbed-os-program"
[mbed] Working path "C:\Users\Ronni\AppData\Local\Temp\mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches... done.
Compiling Mbed Application...pippo
argument -m/--mcu: EDGE_CONTROL is not a supported MCU. Supported MCUs are:
ADV_WISE_1510,          ADV_WISE_1570,          ARCH_MAX,
ARCH_PRO,               ARDUINO_NANO33BLE,      ARM_CM3DS_MPS2,
ARM_MPS2_M0,            ARM_MPS2_M0P,           ARM_MPS2_M3,
ARM_MPS2_M4,            ARM_MPS2_M7,            ARM_MUSCA_B1,
ARM_MUSCA_S1,           B_L4S5I_IOT01A,         B_U585I_IOT02A,
CY8CKIT064B0S2_4343W,   CY8CKIT_062S2_43012,    CY8CKIT_062_BLE,
CY8CKIT_062_WIFI_BT,    CY8CPROTO_062S3_4343W,  CY8CPROTO_062_4343W,
CYSBSYSKIT_01,          CYTFM_064B0S2_4343W,    CYW9P62S1_43012EVB_01,
CYW9P62S1_43438EVB_01,  DISCO_F413ZH,           DISCO_F429ZI,
DISCO_F469NI,           DISCO_F746NG,           DISCO_F769NI,
DISCO_H747I,            DISCO_H747I_CM4,        DISCO_H747I_CM7,
DISCO_L072CZ_LRWAN1,    DISCO_L475VG_IOT01A,    DISCO_L476VG,
DISCO_L496AG,           DISCO_L4R9I,            DISCO_L562QE,
DISCO_WB5MMG,           EFM32GG11_STK3701,      EFM32GG_STK3700,
EP_AGORA,               EP_ATLAS,               EV_COG_AD3029LZ,
EV_COG_AD4050LZ,        FF1705_L151CC,          FF_LPC546XX,
FVP_MPS2_M0,            FVP_MPS2_M0P,           FVP_MPS2_M3,
FVP_MPS2_M4,            FVP_MPS2_M7,            GD32_F307VG,
GD32_F450ZI,            GR_LYCHEE,              GR_MANGO,
HEXIWEAR,               K22F,                   K64F,
K66F,                   K82F,                   KL25Z,
KL43Z,                  KL46Z,                  KW41Z,
LPC1114,                LPC1768,                LPC54114,
LPC546XX,               MAX32620FTHR,           MAX32625MBED,
MAX32625PICO,           MAX32630FTHR,           MAX32660EVSYS,
MAX32670EVKIT,          MIMXRT1050_EVK,         MIMXRT1170_EVK,
MOTE_L152RC,            MTS_DRAGONFLY_F411RE,   MTS_DRAGONFLY_F413RH,
MTS_DRAGONFLY_L471QG,   MTS_DRAGONFLY_L496VG,   MTS_MDOT_F411RE,
NRF52840_DK,            NRF52_DK,               NUCLEO_F070RB,
NUCLEO_F072RB,          NUCLEO_F091RC,          NUCLEO_F103RB,
NUCLEO_F207ZG,          NUCLEO_F303K8,          NUCLEO_F303RE,
NUCLEO_F303ZE,          NUCLEO_F401RE,          NUCLEO_F411RE,
NUCLEO_F412ZG,          NUCLEO_F413ZH,          NUCLEO_F429ZI,
NUCLEO_F439ZI,          NUCLEO_F446RE,          NUCLEO_F446ZE,
NUCLEO_F722ZE,          NUCLEO_F746ZG,          NUCLEO_F756ZG,
NUCLEO_F767ZI,          NUCLEO_G031K8,          NUCLEO_G071RB,
NUCLEO_G0B1RE,          NUCLEO_G431KB,          NUCLEO_G431RB,
NUCLEO_G474RE,          NUCLEO_H743ZI2,         NUCLEO_H7A3ZI_Q,
NUCLEO_L073RZ,          NUCLEO_L152RE,          NUCLEO_L432KC,
NUCLEO_L433RC_P,        NUCLEO_L452RE_P,        NUCLEO_L476RG,
NUCLEO_L486RG,          NUCLEO_L496ZG,          NUCLEO_L496ZG_P,
NUCLEO_L4R5ZI,          NUCLEO_L4R5ZI_P,        NUCLEO_L552ZE_Q,
NUCLEO_U575ZI_Q,        NUCLEO_WB15CC,          NUCLEO_WB55RG,
NUCLEO_WL55JC,          NUMAKER_IOT_M252,       NUMAKER_IOT_M263A,
NUMAKER_IOT_M487,       NUMAKER_PFM_M453,       NUMAKER_PFM_M487,
NUMAKER_PFM_NANO130,    NUMAKER_PFM_NUC472,     NU_M2354,
PORTENTA_H7_M4,         PORTENTA_H7_M7,         RHOMBIO_L476DMW1K,
RZ_A1H,                 S1SBP6A,                S5JS100,
SDP_K1,                 SDT32620B,              SDT32625B,
SDT52832B,              SDT64B,                 SFE_ARTEMIS,
SFE_ARTEMIS_ATP,        SFE_ARTEMIS_DK,         SFE_ARTEMIS_MODULE,
SFE_ARTEMIS_NANO,       SFE_ARTEMIS_THING_PLUS, SFE_EDGE,
SFE_EDGE2,              TB_SENSE_12,            TMPM46B,
TMPM4G9,                TMPM4KN,                UHURU_RAVEN,
WIO_3G,                 WIO_BG96,               WIO_EMW3166,
XDOT_L151CC
[mbed] ERROR: "C:\Users\Ronni\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\python.exe" returned error.
       Code: 2
       Path: "C:\Users\Ronni\AppData\Local\Temp\mbed-os-program"
       Command: "C:\Users\Ronni\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\python.exe -u C:\Users\Ronni\AppData\Local\Temp\mbed-os-program\mbed-os\tools\make.py -t GCC_ARM -m EDGE_CONTROL --source . --build .\BUILD\EDGE_CONTROL\GCC_ARM -v"
---

brinata, 2022-11-29:
Thank you, I'll try.

do I have to clone it or install it using arduino-cli ?
does it exist a kind of guide to setup the building environment under windows ? the script mbed-os-to-arduino recall al lot of unix apps...

manchoz, 2022-11-29:
Hi,
You need to clone it: this is not a core; it is code required to build the [`libmbed.a`](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/EDGE_CONTROL/libs/libmbed.a) library that is part of the core for the Edge Control.

About using `mbed-os-to-arduino` on Windows, please take a look at https://github.com/arduino/ArduinoCore-mbed/issues/570: Cygwin, MinGW, or MSys2 are the way to go.

Please, note `mbed-os-to-arduino` is an internal tool, and it is only required to build the `libmbed.a` library for the board variants after you make changes to the base MbedOS code. It is not meant to build MbedOS applications for Arduino boards.

brinata, 2022-11-29:
I've received this erros:

Copying generic MbedOS headers to core... ./mbed-os-to-arduino: line 252: rsync: command not found

I'm working under the git bash console but I'm not able to find rsync for windows...

manchoz, 2022-11-29:
There are several ports of RSync for Windows, depending on your scenario.

Standalone:
* [Scoop](https://scoop.sh) - https://scoop.sh/#/apps?q=rsync&s=0&d=1&o=true
* [Choco](https://chocolatey.org) - https://community.chocolatey.org/packages?q=rsync 

Emulation Environment:
* [CygWin](https://www.cygwin.com/index.html) - https://www.cygwin.com/cgi-bin2/package-grep.cgi?grep=rsync
* [MSys2](https://www.msys2.org) - https://packages.msys2.org/search?q=rsync