arduino/ArduinoCore-mbed #600: Can't Upload WiFi Firmware

alwint3r, 2022-12-24:
I wanted to use the WiFi on a freshly opened Nicla Vision board, so I had to update the firmware first. But I encountered an issue where the WiFi partition couldn't be formatted. I used the v3.4.1 version from the board manager. I added a few check codes to check whether the format process results in an error or not, and here's the output:

```
root::init result: 0
No filesystem containing the WiFi firmware was found.
Usually that means that the WiFi firmware has not been installed yet or was overwritten with another firmware.

Formatting the filsystem to install the firmware and certificates...

Format result: -3101
Is fp null? 1
Flashing /wlan/4343WA1.BIN file
Flashed 0%
Writing
```

After an hour of pulling my long hair, I had to switch to my own setup which HEAD points to 933a3e143adb9d0dca13515800ad02261a7675eb commit. Then it worked.

facchinm, 2023-01-02:
Hi @alwint3r , sorry for the inconvenience. I found the bug and pushed a fix (on mbed-os tree), it should be picked up in the next core release (likely in a few days).
When everything will be ok I'll close the issue properly.

Thanks

alwint3r, 2023-01-21:
I updated the core version to 3.5.4, and I can confirm that I can write the WiFi firmware to the Nicla Vision board.

I'm closing this issue.

Thanks!

closed 2023-01-21 by alwint3r