arduino/ArduinoCore-mbed #601: Should `arduino::MbedI2C::read()` return `-1` if there is no data available?

carlosperate, 2022-12-25:
Assuming https://github.com/arduino/reference-en/issues/897 in the docs repo is correct, it looks like `arduino::MbedI2C::read()` should be returning `-1` instead of `0` at the end of the method:
https://github.com/arduino/ArduinoCore-mbed/blob/949c70ce6ae7d2910e5768d6d88fc3ae0e943584/libraries/Wire/Wire.cpp#L119-L124

As reference, these implementations return `-1`:
- Any ArduinoCore using the ArduinoCore-API RingBuffer for the rx buffer: https://github.com/arduino/ArduinoCore-API/blob/master/api/RingBuffer.h#L96
- ArduinoCore-sam: https://github.com/arduino/ArduinoCore-sam/blob/master/libraries/Wire/src/Wire.cpp#L289
- ArduinoCore-avr: https://github.com/arduino/ArduinoCore-avr/blob/42fa4a1ea1b1b11d1cc0a60298e529d37f9d14bd/libraries/Wire/src/Wire.cpp#L293

facchinm, 2023-01-02:
Hi @carlosperate , you are very right indeed :slightly_smiling_face: 
Would you mind submitting a PR so we can give proper attribution for the fix? Thx

carlosperate, 2023-01-02:
Sure thing! I'll create one between tonight and tomorrow 👍

closed 2023-01-03 by facchinm