arduino/ArduinoCore-mbed #602: Feature Request: the inclusion of professionally constructed websocket server and client WiFi and Ethernet examples in Arduino Pro.

hpssjellis, 2022-12-28:
**Feature Request: the inclusion of professionally constructed websocket server and client WiFi and Ethernet examples in Arduino Pro.** 

As the use of microcontrollers in IoT continues to grow, there is an increasing demand for fast and secure data transfer both over the internet and within local networks. Providing websocket examples in Arduino Pro would allow users to more easily implement these features in their projects and meet the growing demand for secure and fast data transfer in IoT systems. (Generated by ChatGPT)

I have already hacked together some forms of solutions to at least the WiFiSSLClient side of websockets using the Arduino Portenta in these 2 closed issues:
  
https://github.com/arduino/ArduinoCore-mbed/issues/599
https://github.com/arduino/ArduinoCore-mbed/issues/589

but my solutions have a few minor issues which Arduino could tidy. I would be willing to help where I can.

hpssjellis, 2023-01-04:
I am going to leave this issue open, as I do not know how to make a websocket server, but I have got working the WiFiSSLclient as a websocket. I had to actually do a fair bit of the work to get it going. My "Portenta Pro Community Solutions" library release version 0.9.3 has a few examples of working websockets in the ```dot36``` area. 

The best example is probably [dot364 here](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot3-portenta-vision-shields/dot36-camera-png-to-web/dot364-camera-activated-websocket-to-sd-card/dot364-camera-activated-websocket-to-sd-card.ino).

hpssjellis, 2023-01-05:
@facchinm Arduino MBED doesn't seem to have a WiFiSSLServer library in the WiFi library at 

https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/WiFi/src or in the socketWrapper libarary at 

https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/SocketWrapper/src

does that mean there is no point my trying to make a websocket SSL server at the moment? I am reasonably happy with my WiFiSSLClient websocket examples at 

https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot36-camera-png-to-web

facchinm, 2023-01-05:
@hpssjellis acting as an SSL server is not trivial, especially setting up the CA chain. Since all the infrastructure is based on mbedtls, porting something like https://github.com/Mbed-TLS/mbedtls/blob/master/programs/ssl/ssl_server.c should be feasible, but the default wrapper (TLSSockerWrapper) can't be used out of the box (`listen()` and `accept()` are not implemented https://os.mbed.com/docs/mbed-os/v6.15/apis/secure-socket.html )