arduino/ArduinoCore-mbed #609: Need help compiling mbed for PORTENTA_H7_M7

salasidis, 2023-01-21:
Hello, having eternal issues trying to compile new libmbed.a file

Created a new PC with Ubuntuu linux
Downloaded Arduino IDE 2.0
Install Portenta Board manager

Used the main file below as guide

https://github.com/arduino/ArduinoCore-mbed#mbed-os-to-arduino-script

Did not create a new clone, but trying to compile in the 3.5.4 installation itself
Went to ~/.arduino15/packages/arduino/hardware/mbed_portenta/3.5.4

I tried running the mbed-os-to-arduino script at that point, and was getting a line 25 error

I then found this
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/66

I installed python3
Created a virtual environemt and and ran 

`./mbed-os-to-arduino -a -g PORTENTA_H7_M7:PORTENTA_H7_M7`

I then got an error message later in the script

`./mbed-os-to-arduino -a -g PORTENTA_H7_M7:PORTENTA_H7_M7
MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=1
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/home/rsalasidis/.arduino15/packages/arduino/hardware/mbed_portenta/3.5.4

VARIANT=PORTENTA_H7_M7 BOARD=PORTENTA_H7_M7 Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" 
(program)
[mbed] PORTENTA_H7_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program) [mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
  done.
Applying patches... done.
Restoring gdb info (this increases libmbed binary size, not suitable for
release)
Updated 0 paths from the index
/tmp/mbed-os-program
Compiling Mbed Application...[mbed] ERROR: 
"/home/rsalasidis/.arduino15/packages/arduino/hardware/mbed_portenta/3.5.4/bin/python3" 
returned error.
        Code: 1
        Path: "/tmp/mbed-os-program"
        Command: 
"/home/rsalasidis/.arduino15/packages/arduino/hardware/mbed_portenta/3.5.4/bin/python3
-m pip list -l"
---

`

I downloaded the GNU ToolChain, but not sure if the problem is that the scrtipt dows not find the compiler locations?

Any ideas?

Thanks

salasidis, 2023-01-21:
I did the clone, and created the symbolic link

Everything else was left as is

I now get the following error message ? compiler location ?

```
./mbed-os-to-arduino -a -g PORTENTA_H7_M7:PORTENTA_H7_M7

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=1
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/home/rsalasidis/Arduino/hardware/arduino-git/mbed

VARIANT=PORTENTA_H7_M7 BOARD=PORTENTA_H7_M7
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] PORTENTA_H7_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches...git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0001-PACKAGING-remove-gdb-exta-information-to-slim-librar.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0002-Portenta-enable-certificates-in-filesystem.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0003-I2CSlave.read-buf-count-returns-the-number-of-bytes-.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0004-Add-AnalogIn-configure-function.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0005-HACK-avoid-10862-by-not-firing-the-assert.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0006-portentaH7-Change-BLE-public-address.patch 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0006-portentaH7-Change-BLE-public-address.patch:21: trailing whitespace.
            uint8_t bdAddr[6] = { 0xFF, 0xFF, 0xFF, 0x0A, 0x61, 0xA8 }; 
warning: 1 line adds whitespace errors.
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0007-TLSSocketWrapper-add-method-to-retrieve-certificates.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0008-TLSSocketWrapper-Add-unit-test-for-set_root_ca_cert_.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0009-TLSSocketWrapper-add-test-for-certificates-stored-in.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0010-Initial-support-Portenta-Beta.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0011-STM32-USB-allow-dynamic-USB-speed-configuration-base.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0012-Increase-Cordio-memory-pool-for-nRF52832.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0013-Cordio-Add-sendCmplt-Callback-for-updating-NumCmplPc.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch:139: trailing whitespace.
    
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch:182: trailing whitespace.
    
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch:197: trailing whitespace.
    QSPI_FLASH1_CSN = QSPI1_CSN,    
warning: 3 lines add whitespace errors.
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch:1045: trailing whitespace.
by the actual target `PICO_PLATFORM`` 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch:3450: trailing whitespace.
 *         // will try to sleep until timeout or the next processor event 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch:5113: trailing whitespace.
This can be extremely useful for testing and debugging higher level application code, or porting code which is not yet small enough 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch:5119: trailing whitespace.
It is possible however to inject additional SDK library implementations/simulations to provide 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0015-RASPBERRYPI-import-pico-sdk.patch:5120: trailing whitespace.
more complete functionality. For an example of this see the [pico-host-sdl](https://github.com/raspberrypi/pico-host-sdl) 
warning: squelched 50 whitespace errors
warning: 55 lines add whitespace errors.
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0016-RP2040-add-NANO_RP2040_CONNECT-target.patch 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0016-RP2040-add-NANO_RP2040_CONNECT-target.patch:626: new blank line at EOF.
+
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0016-RP2040-add-NANO_RP2040_CONNECT-target.patch:820: new blank line at EOF.
+
warning: 2 lines add whitespace errors.
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0017-RP2040-add-placeholder-files-for-includes-like-pico-.patch 
[... 271 lines cut ...]
/tmp/mbed-os-program
Compiling Mbed Application...[mbed] WARNING: Missing Python modules were not auto-installed.
[... cut ...]
ModuleNotFoundError: No module named 'jinja2'
[mbed] ERROR: "/home/rsalasidis/Arduino/hardware/arduino-git/mbed/virtenv/bin/python3" returned error.
       Code: 1
       Path: "/tmp/mbed-os-program"
       Command: "/home/rsalasidis/Arduino/hardware/arduino-git/mbed/virtenv/bin/python3 -u /tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m PORTENTA_H7_M7 --source . --build ./BUILD/PORTENTA_H7_M7/GCC_ARM -v"
---
(virtenv) rsalasidis@rsalasidis-Aspire-S7-391:~/Arduino/hardware/arduino-git/mbed$
```

trik, 2023-01-21:
@salasidis some python modules are missing. use this requirements file to install all the dependencies https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt

salasidis, 2023-01-21:
Thanks, I see that now in the list. I have once upon a time used QNX about 25 yrs ago, but have not used any UNIX/Linux flavor since, so I am quite rusty - sorry.

I managed to install them - had to use 'sudo pip install -r requirements.txt'

Interestingly enough, it worked  the first time, but if I try to install again, I get all sorts of errors - below is a short list of the start of the output

```
$ sudo pip install -r /home/Downloads/requirements.txt
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    load_entry_point('pip==20.0.2', 'console_scripts', 'pip')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/cmdoptions.py", line 24, in <module>
    from pip._internal.exceptions import CommandError
```

In any case, when I try to run the ./mbed-os-to-arduino script again I also get a variable output - after running a few times, I now get this

```
$ ./mbed-os-to-arduino -a -g PORTENTA_H7_M7:PORTENTA_H7_M7

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=1
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/home/rsalasidis/Arduino/hardware/arduino-git/mbed

VARIANT=PORTENTA_H7_M7 BOARD=PORTENTA_H7_M7
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] PORTENTA_H7_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches...git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0001-PACKAGING-remove-gdb-exta-information-to-slim-librar.patch 
error: patch failed: tools/profiles/develop.json:5
error: tools/profiles/develop.json: patch does not apply
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0002-Portenta-enable-certificates-in-filesystem.patch 
error: patch failed: targets/targets.json:3428
error: targets/targets.json: patch does not apply
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0003-I2CSlave.read-buf-count-returns-the-number-of-bytes-.patch 
error: patch failed: drivers/include/drivers/I2CSlave.h:172
error: drivers/include/drivers/I2CSlave.h: patch does not apply
error: patch failed: drivers/source/I2CSlave.cpp:52
error: drivers/source/I2CSlave.cpp: patch does not apply
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0004-Add-AnalogIn-configure-function.patch 
error: patch failed: drivers/include/drivers/AnalogIn.h:88
error: drivers/include/drivers/AnalogIn.h: patch does not apply
error: patch failed: hal/include/hal/analogin_api.h:33
error: hal/include/hal/analogin_api.h: patch does not apply
error: patch failed: targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c:122
error: targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c: patch does not apply
error: patch failed: targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/objects.h:50
[... 19 lines cut ...]
error: patch failed: connectivity/netsocket/tests/UNITTESTS/netsocket/TLSSocketWrapper/test_TLSSocketWrapper.cpp:399
error: connectivity/netsocket/tests/UNITTESTS/netsocket/TLSSocketWrapper/test_TLSSocketWrapper.cpp: patch does not apply
error: patch failed: connectivity/netsocket/tests/UNITTESTS/netsocket/TLSSocketWrapper/tls_test_config.h:19
error: connectivity/netsocket/tests/UNITTESTS/netsocket/TLSSocketWrapper/tls_test_config.h: patch does not apply
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0009-TLSSocketWrapper-add-test-for-certificates-stored-in.patch 
error: patch failed: connectivity/netsocket/tests/TESTS/netsocket/tls/main.cpp:234
error: connectivity/netsocket/tests/TESTS/netsocket/tls/main.cpp: patch does not apply
error: patch failed: connectivity/netsocket/tests/TESTS/netsocket/tls/tls_tests.h:89
error: connectivity/netsocket/tests/TESTS/netsocket/tls/tls_tests.h: patch does not apply
error: connectivity/netsocket/tests/TESTS/netsocket/tls/tlssocket_cert_in_filesystem.cpp: already exists in working directory
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0010-Initial-support-Portenta-Beta.patch 
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0011-STM32-USB-allow-dynamic-USB-speed-configuration-base.patch 
error: patch failed: targets/TARGET_STM/USBPhy_STM32.cpp:217
[... cut ...]
error: connectivity/FEATURE_BLE/libraries/TARGET_CORDIO_LL/stack/controller/sources/ble/lctr/lctr_main_conn.c: patch does not apply
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch 
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch:139: trailing whitespace.
    
/home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0014-Add-support-for-Arduino-Edge-Control.patch:182: trailing whitespace.
```

salasidis, 2023-01-21:
This is the content of the 0001-PACKAGING...   file
```
From 3d7f8d18abca4e6132611fe544bf94b5d3483af3 Mon Sep 17 00:00:00 2001
From: Martino Facchin <m.facchin@arduino.cc>
Date: Thu, 11 Mar 2021 15:05:20 +0100
Subject: [PATCH 001/157] PACKAGING: remove gdb exta information to slim
 library

---
 tools/profiles/develop.json | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/tools/profiles/develop.json b/tools/profiles/develop.json
index 21ba5c88d5..5995d276e4 100644
--- a/tools/profiles/develop.json
+++ b/tools/profiles/develop.json
@@ -5,7 +5,7 @@
                    "-fmessage-length=0", "-fno-exceptions",
                    "-ffunction-sections", "-fdata-sections", "-funsigned-char",
                    "-MMD",
-                   "-fomit-frame-pointer", "-Os", "-g", "-DMBED_TRAP_ERRORS_ENABLED=1"],
+                   "-fomit-frame-pointer", "-Os", "-DMBED_TRAP_ERRORS_ENABLED=1"],
         "asm": ["-c", "-x", "assembler-with-cpp"],
         "c": ["-c", "-std=gnu11"],
         "cxx": ["-c", "-std=gnu++14", "-fno-rtti", "-Wvla"],
-- 
2.36.0
```

I 
`cd to /tmp/mbed-os-program/mbed-os`

In ./tools/profiles from that directory the devbelop.json file is there

These are the files present in that directory
`debug.json  develop.json  extensions  release.json`

When I execute the git apply command, I get the following

```
git apply /home/rsalasidis/Arduino/hardware/arduino-git/mbed/patches/0001-PACKAGING-remove-gdb-exta-information-to-slim-librar.patch
error: patch failed: tools/profiles/develop.json:5
error: tools/profiles/develop.json: patch does not apply
```

No difference if I execute it as sudo

Still lost unfortunately - Thanks for any further help

facchinm, 2023-01-23:
Hi @salasidis ,
I think the missing dependencies messed up with the script a bit.
Before retrying running `mbed-os-to-arduino`, make sure to delete `/tmp/mbed-os-program` folder, so the next run will not use the cached version of mbed-os but will clone a new one and likely checkout `latest` branch (which is the only one where the patches apply correctly).

About bundling the patches with the released version of the core there's an issue already opened here https://github.com/arduino/ArduinoCore-mbed/issues/587 that may contains some information to get you unstuck

salasidis, 2023-01-23:
Thanks for taking the time. That did help in getting the compiler to continue running again. There were some more dependencies and the like I had to install, but it eventtually says 

`[mbed-os-to-arduino] MbedOS core generation ended succesfully.`

In case others are interested, I created a file that has the steps I took in order to get it to work

```
•	Install wsl2 ubuntu and desktop

•	https://medium.com/@riley.kao/wsl2-ubuntu20-04-gui-remote-desktop-connection-rdp-2bbd21d2fa71

•	Install Arduino IDE via image from Arduino.cc

•	Start xrdp

o	sudo /etc/init.d/xrdp start

# check xrdp status (optional)
sudo /etc/init.d/xrdp status

•	Create virtual environment
o	[How to Create and Activate Virtual Environment In Linux ? | by Baleshwar Mahto | Medium](https://baleshwar-mahto.medium.com/how-to-create-and-activate-virtual-environment-in-linux-b3c68d4d60fb)
o	source bin/activate

•	Add ssh key to github
o	https://www.theserverside.com/blog/Coffee-Talk-Java-News-Stories-and-Opinions/GitHub-SSH-Key-Setup-Config-Ubuntu-Linux

•	add requirements in new dependency
o	https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt
o	pip install -r requirements.txt

•	install npm xpm
o	install npm – sudo apt install npm
o	sudo npm install –global xpm@latest

•	Install Compilers
o	sudo apt install libc6-armel-cross libc6-dev-armel-cross binutils-arm-linux-gnueabi libncurses5-dev build-essential bison flex libssl-dev bc
o	(not sure above one is necessary – but that is the order I did it)
o	sudo apt-get install gcc-arm*

•	Install mbed
o	[Install and set up - Build tools | Mbed OS 6 Documentation](https://os.mbed.com/docs/mbed-os/v6.15/build-tools/install-and-set-up.html)

•	Install iq
o	sudo apt install jq 

•	Install steps from [arduino/ArduinoCore-mbed (github.com)](https://github.com/arduino/ArduinoCore-mbed#mbed-os-to-arduino-script)
o	mkdir -p $sketchbook/hardware/arduino-git
o	cd $sketchbook/hardware/arduino-git
o	git clone git@github.com:arduino/ArduinoCore-mbed mbed
o	-
o	git clone git@github.com:arduino/ArduinoCore-API
o	–
o	Make a symbolic link in the mbed directory of the ArduinoCore-API/api directory
	Easiest to use GUI and Shift/Ctrl and drag one directory to the other

o	Go to mbed directory 
o	cd $sketchbook/hardware/arduino-git/mbed
o	./mbed-os-to-arduino -a -g PORTENTA_H7_M7:PORTENTA_H7_M7

remember to 'sudo rm -r -f /tmp/mbed-os-program' between every compile (or compile attempt)

And with any luck, that should work
```