arduino/ArduinoCore-mbed #610: Arduino Nano RP2040 Connect triggers notification every time I upload a new sketch on MacOS

LazaroFilm, 2023-01-21:
The following notification pops up every time I upload a sketch to the Nano RP2040 Connect. Worse, they are persistent, and they stack up into a column on the side of my screen until I close each one individually.

[image](https://user-images.githubusercontent.com/18351100/213880382-16e644d3-6ee9-4dd3-a2d5-633d50046d1a.jpeg)

This is likely due to the way RP2040 works by opening a drive to transfer the compiled file then resets, cutting the connection without properly ejecting. Maybe there would be a way to add a command to eject the drive prior to reset? 
I'm using Arduino IDE 2.0.3

facchinm, 2023-01-23:
Hi @LazaroFilm 
since our upload strategy for rp2040 uses `picoboot` interface and Mass storage is not needed we could "simply" change https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/NANO_RP2040_CONNECT/variant.cpp#L94 last parameter to 1 (see https://github.com/arduino/ArduinoCore-mbed/blob/740d4aca0d2862451355515fa00956f16533efce/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_bootrom/include/pico/bootrom.h#L146-L149 

Unfortunately, if we apply this change, the board will stop working on Windows due to drivers incompatibilities, so we can't do it for everyone.
But if you are sure you'll never use the board on Windows feel free to change that line and get rid of the notifications :slightly_smiling_face:

LazaroFilm, 2023-01-23:
Stop a big AND make it incompatible with windows‽ that's two birds with one stone! 
I'll definitely do that. If I need to have it work with Windows, I can always re-flash it with the 0 value.