arduino/ArduinoCore-mbed #615: [TC-440] Blink_Nicla.ino compiles for Nicla Sense, but not Nicla Vision

aliphys, 2023-01-30:
**Background info**

According to https://github.com/arduino/arduino-examples/issues/47 there is a special Blink.ino for the Nicle Sense ME, which according to https://docs.arduino.cc/software/ide-v1/tutorials/getting-started/cores/arduino-mbed_nicla also applies to the Nicla Vision. This is called `Blink_Nicla.ino`.

It can be found in this repo here: https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/Nicla_System/examples/Blink_Nicla/Blink_Nicla.ino

**Expected behaviour**
I can compile the `Blink_Nicla.ino` sketch to both the Nicla Sense ME as well as the Nicla Vision

**Current Behaviour**
When the Nicla Sense ME is selected, IDE 2 compiles without error.
[image]
However, when compiling for the Nicla Vision, it fails to compile.
[image]

Error message is as below
```
In file included from C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\libraries\Nicla_System\src/Nicla_System.h:6:0,
                 from C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2023030-12348-k6zwi4.8tz8\Blink_Nicla\Blink_Nicla.ino:1:
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\libraries\Nicla_System\src/RGBled.h:91:14: error: expected identifier before numeric constant
 #define Mode 1
              ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\cores\arduino/mbed/drivers/include/drivers/interfaces/InterfaceCAN.h:150:10: note: in expansion of macro 'Mode'
     enum Mode {
          ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\libraries\Nicla_System\src/RGBled.h:91:14: error: expected unqualified-id before numeric constant
 #define Mode 1
              ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\cores\arduino/mbed/drivers/include/drivers/interfaces/InterfaceCAN.h:150:10: note: in expansion of macro 'Mode'
     enum Mode {
          ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\libraries\Nicla_System\src/RGBled.h:91:14: error: expected identifier before numeric constant
 #define Mode 1
              ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\cores\arduino/mbed/drivers/include/drivers/CAN.h:171:14: note: in expansion of macro 'Mode'
     int mode(Mode mode);
              ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\libraries\Nicla_System\src/RGBled.h:91:14: error: expected ',' or '...' before numeric constant
 #define Mode 1
              ^
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nicla\3.5.1\cores\arduino/mbed/drivers/include/drivers/CAN.h:171:14: note: in expansion of macro 'Mode'
     int mode(Mode mode);
              ^

exit status 1

Compilation error: exit status 1
```

facchinm, 2023-01-31:
Nicla Vision doesn't have the LED on external I2C controller, so it can use the normal blink. Best thing we can do is adding an `#error` directive when compiling `Nicla_System` library for Nicla Vision, so the error is more precise.

aliphys, 2023-02-02:
@facchinm So something like this?
```
#ifdef ARDUINO_NICLA_VISION
  #error "Run the standard Blink.ino sketch for the Nicla Vision"
```

facchinm, 2023-02-02:
yup

deemano, 2023-02-10:
> Nicla Vision doesn't have the LED on external I2C controller, so it can use the normal blink. Best thing we can do is adding an `#error` directive when compiling `Nicla_System` library for Nicla Vision, so the error is more precise.

So, basically, there is no Blink sketch for Nicla Vision. It's that what you're saying?

Q: It's there any reason why compiling is very slow for Vision? [using the latest Arduino IDE]