arduino/ArduinoCore-mbed #617: EthernetServer inconsistencies

KimCranfield, 2023-02-02:
Trying to create a simple TCP server on a Portenta Machine control board.    To support a single ocasional connection for a configuration interface.    Board will run a main loop, with spare time used to poll, parse etc.

I started with the ChatServer example, but that closes the connection as soon as it opens, after first byte transferred.

WebServer example seems to work, but any attempt to move the polling of _server.available_  out of a while loop into the main loop makes it drop the connection.

This works - if after the connection I _immediately_ drop into a while loop to poll for input.   Telnet in, connection stays open, bytes sent both ways.    Interestingly _client.available_ doesn't return true until there is a newline, but I can work with that.

The printing on no input is just to show activity while debugging.
```
void loop() {
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        client.write(c);
      }else {
        Serial.print("n");
        delay(50);
      }
    }
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```

But if I restructure to just add a flag and then drop into the while loop if that flag is true (which it is!) it drops the connection immediately after it is opened (Telnet from Linux)

My plan for this flag was to move from a local while loop to an if, calling it from the main loop so the rest of the system keeps running - but even before I got to that, leaving the local while loop, it dies.
```
void loop() {
  if (not(flagConnected)) {
    // listen for incoming clients
    EthernetClient client = server.available();
    if (client) {
      Serial.println("new client");
      flagConnected=true;
    }
  }

  if (flagConnected) {   
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        client.write(c);
      }else {
        Serial.print("n");
        delay(50);
      }
    }
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
    flagConnected=false;
  }
  
}
``` 
Using mbed_portenta library from the library manager, v3.5.4  Arduino IDE 1.8.19    Portenta Machine Control board, with H7 and using the M7 core.

facchinm, 2023-02-02:
Hi @KimCranfield ,
you are shadowing the global `client` object with the one that you get from `server.available();`  
```
  if (not(flagConnected)) {
    // listen for incoming clients
    EthernetClient client = server.available();   <--- this creates a new object
    if (client) {
      Serial.println("new client");
      flagConnected=true;
    }
  } <---- that gets destroyed in this line (when it goes out of scope, closing the connection)
```

The correct code could resemble
```

closed 2023-02-02 by facchinm

KimCranfield, 2023-02-23:
Thank you ! - rookie error.   
Sorry for the delay - I did reply that afternoon but I don't think it stuck.