arduino/ArduinoCore-mbed #619: Low power mode for Portenta system

TuringTW, 2023-02-04:
Hi, I want to ask if the low-power mode of Portenta H7 is supported using Arduino code. I read through the repo and found that the [enterLowPower ](https://github.com/arduino/ArduinoCore-mbed/blob/933a3e143adb9d0dca13515800ad02261a7675eb/libraries/STM32H747_System/src/Portenta_System.cpp)function of the Portenta system still needs to be implemented. I am wondering if there's any plan to implement it or any example I could start with.

hpssjellis, 2023-02-05:
I find the Portenta to be a bit of a power hog, even on a USB battery pack running WiFi it doesn't seem to last very long.

I would also be really interested in a Low-Power-Mode but perhaps an easier question is, if there are ways to shut down specific chips. On the Portenta, can we shutdown the Murata BLE/WiFi chip, the SDRAM, the cyrpto chip or anything else that uses standby power? Also on the vision shields can we shut down: Ethernet, the Murata LoRa Modula, the microphones or the always-on grayscale camera?

I think the answer is "No" but it is always a good idea to ask.

TuringTW, 2023-02-05:
@hpssjellis I found that there's an implementation of the [enterLowPower ](https://github.com/arduino/ArduinoCore-mbed/blob/933a3e143adb9d0dca13515800ad02261a7675eb/libraries/STM32H747_System/src/Nicla_Vision_System.cpp)function of Nicla vision system, which share the same main controller and probably some other chips. In that function, it turns off several chips and use the internal clock of the main chip. It might give us some hints on it.

hpssjellis, 2023-02-06:
@TuringTW that is really interesting! Look at the code I just wrote to reset the PMIC using I2C to solve the [orange Led of Hell](https://github.com/arduino/ArduinoCore-mbed/issues/608#issuecomment-1413178638) issue

```
        writeI2cRocksetta(devAddr, 0x4f, 0x0);// LDO2 to 1.8V //initialize the PMIC registers:
        writeI2cRocksetta(devAddr, 0x4c, 0xf);// LDO1 to 1.0V
        writeI2cRocksetta(devAddr, 0x4d, 0xf);
        writeI2cRocksetta(devAddr, 0x52, 0x9);// LDO3 to 1.2V
        writeI2cRocksetta(devAddr, 0x53, 0xf);
        osDelay(10);
        writeI2cRocksetta(devAddr, 0x9c, (1 << 7) );//0x80 //charger LED off - duty cycle
        writeI2cRocksetta(devAddr, 0x9e, (1 << 5) );//0x20 // Disable charger led
        osDelay(10);
        writeI2cRocksetta(devAddr, 0x42, 2);    // SW3: set 2A as current limit // Helps keeping the rail up at wifi startup
        osDelay(10);
        writeI2cRocksetta(devAddr, 0x94, (20 << 3) );//0xA0 // Change VBUS INPUT CURRENT LIMIT to 1.5A
        writeI2cRocksetta(devAddr, 0x3b, 0xf);// SW2 to 3.3V (SW2_VOLT)

```

and the code you just showed me for the Nicla Vision. The values don't match up but the programming is really interesting.

```
void arduino::Nicla_Vision_System::enable3V3VDDIO_EXT() {
  //Power SW2 off
  STM32H747::setRegister(0x3B,0x80);
  delay(10);
  //Set SW2 to 3.3V
  STM32H747::setRegister(0x38,0x7);
  //Power SW2 on
  STM32H747::setRegister(0x3B,0x0F);
}

void arduino::Nicla_Vision_System::enable1V8VDDIO_EXT() {
  //Power SW2 off
  STM32H747::setRegister(0x3B,0x80);
  delay(10);
  //Set SW2 to 1.8V
  STM32H747::setRegister(0x38,0x4);
  //Power SW2 on
  STM32H747::setRegister(0x3B,0x0F);
}

void arduino::Nicla_Vision_System::disableVDDIO_EXT() {
  //Power SW2 off
  STM32H747::setRegister(0x3B,0x80);
}
```

My code  [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/research/i2c-orange/a12-full-app-orange-led-fix.ino) lets us look at the first 160 registers at the PMIC bootloader start, but changing those values is very dangerous. I just bricked a Portenta last week! It would be nice if an Arduino employee would add an opinion here. Given the correct values to turn power on or off we could probably write the low power code ourselves.

Question: For anyone @facchinm or others.

does ```  STM32H747::setRegister(0x3B,0x80);```  permanently change the register or on next boot does the register go back to what it was originally set to?

The best forum thread about the Portenta low power is probably [here](https://forum.arduino.cc/t/portenta-h7-low-power-mode-project/1085322)

facchinm, 2023-02-06:
Hi @TuringTW and Jeremy,
in fact we are working on completing the low power implementation for boards based on H747.
The topic is not trivial, since the M4 part MUST run some proper code to allow the system to go into low power properly.
Some extra hardware, like ethernet phy or usb phy, can be turned off via carefully writing PMIC registers (that will reset at default value as soon as you unplug the power, to answer @hpssjellis ) or can be set into low power mode via esoteric methods (like [this](https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Projects/STM32F769I-Discovery/Examples/PWR/PWR_CurrentConsumption/Src/stm32f7xx_lp_modes.c#L654-L657) undocumented stuff).
In the meantime, if you want to experiment, please not that the board can always be recovered by:
* disconnecting the power
* reconnecting while pressing the reset button
* release the reset and press it again immediately (a sort of cold double tap)

This will allow reentering bootloader mode before the sketch is executed, so a known good sketch can be reflashed.

hpssjellis, 2023-02-06:
@TuringTW I now have a reason to learn how to use my Nordic Power Profiler PPK2.  Fairly sure I didn't not use the Nordic nrf-connect power profiler software when I first tried and nothing worked. This is much more interesting now.

[image]

Anyway, here is the startup power running into the Portenta VIN pin for the first 2 seconds. Fairly safe to say the different power surges are turning on different chips.  

[image]

And here is just running a hello-blink code that loads an LED and prints to the serial monitor (not connected)

[image]

We should be able to shutdown different chips and see if they have any effect.

Here is output or about 30 seconds for WiFi to make a connection. (Note: my WiFi delays about 15 s just to allow you to connect the monitor if you want)

[image]

Here is a screenshot running osDelay, thread_sleep_for and delay. All don't seem to have any effect.

[image]

actual code is here

```
#include <Arduino.h> // Only needed by https://platformio.org/

#include "mbed.h"
using namespace mbed;

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);   // try on Portenta LEDB = blue, LEDG or LED_BUILTIN = green, LEDR = red 
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW); // turns onboard LED on
  osDelay(500);
  thread_sleep_for(300);
  digitalWrite(LED_BUILTIN, HIGH); // turns onboard LED on
  delay(1000);            
}

```

It should be a lot of fun to mess with the power startup. If anyone can find reference material so we kind of know what we are doing that would be nice. Basically I just want to be able to shutdown specific chips when I know that I am not using them. A generic lowPower mode is not very useful, because I am sure I will need one of the things shutdown.

hpssjellis, 2023-02-07:
The Nicla Vision is definitely running at lower power than the Portenta at around 50 mA instead of about 110 mA. one confusing part is that the power consumption does not seem to drop when I enterLowPower at about 20 seconds into the code.
```
  bool myLow = Nicla_Vision.enterLowPower();
  Serial.print("Nicla_Vision.enterLowPower();  :");
  Serial.println(myLow);
```
and running the above code seems to kill the sketch. Is there a way to exitLowPower?

[image]

I did manage to get a printout of what I think is the PMIC area on the Nicla Vision using my code [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/research/i2c-orange/c05-nicla-low-power-read-PMIC.ino)
```
0x7c, 0x00, 0x11, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x07, 0x00, 0x00, 0x07, 0x00, 0x00, 0x03,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x05, 0x00, 0x00, 0x00, 0x3f, 0x00, 0x00, 0x08, 0x1f, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x04, 0x04, 0x04, 0x03, 0x00, 0x00, 0x07, 0x07, 0x07, 0x03, 0x00, 0x00, 0x0f, 0x0f,
0x0f, 0x03, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1a, 0x03, 0x00, 0x00,
0x03, 0x00, 0x10, 0x03, 0x00, 0x00, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x01, 0x03, 0x04, 0x05, 0x06, 0x00, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x68, 0x00, 0xff, 0x00, 0xac, 0x00, 0x20, 0x03, 0x04, 0x02, 0x00, 0x00, 0x00, 0x40, 0x03, 0x2b,
0x00, 0x00, 0x00, 0x00, 0xa0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x20, 0x00,
```

One nice thing about the above code is that no extra Portenta is needed to read these registers, like when I was trying to solve the Orange LED issue.

One thing I am finding really confusing is that the PMIC address of 0x08 is exactly the same as the first I2C address, so my code seems to be working but I am not sure what address is important when reading and writing registers. I think it is the PMIC address location, but then I am not sure how the code decides which I2C channel to use.   

wire0, wire1, wire2 probably help decide which I2C to use.

hpssjellis, 2023-05-31:
@TuringTW I have never had much luck with lowPower mode on the Portenta. I think Arduino is working on it. I do think it is easier to do all things low power when using an external LiPo 3.7 V battery. Which I have never found a battery I can purchase with the correct 3 prong connector (Can monitor temperature). I don't feel like purchasing the battery and connector separately.  Hopefully this thread will alert us to any changes to the low power software.   https://github.com/arduino/ArduinoCore-mbed/issues/619#issuecomment-1418810715 . I just saw your question there, no I don't think the Cube IDE will help, but you never know.  

I just saw this from that other post. I don't think I have ever done it, but need to try it with a few of my dead Portenta's    

```
In the meantime, if you want to experiment, please note that the board can always be recovered by:

disconnecting the power
reconnecting while pressing the reset button
release the reset and press it again immediately (a sort of cold double tap)
This will allow reentering bootloader mode before the sketch is executed, so a known good sketch can be reflashed.
```

TuringTW, 2023-06-13:
@hpssjellis Hi Jeremy, I just noticed that you pointed out that there's no way to exit the low-power mode for Nicla_Vision. Have you been able to find a solution to exit the mode? 
Another question is that both Nicla_vision and portenta use chips in the same series STM32H747. Is it possible to adapt enterLowerPower for Nicla_vision to portenta?

hpssjellis, 2023-06-13:
@TuringTW I have moved on to other issues, as I think Arduino is working on getting low power working on these devices.

Sarah26-10, 2023-08-22:
Did something happen here?
I want to reduce the power of my Portenta H7 Machine control in some time ranges when nothing happens and we are waiting for a defined temperature (which is given by an I2C connection). 

:)

facchinm, 2023-09-16:
@Sarah26-10 I added some information on "cross-platform" low power [here](https://github.com/arduino/ArduinoCore-mbed/issues/713#issuecomment-1722243089) ; however, Portenta H7 is a complex beast and we are currently writing a proper LowPower + Battery library that will help getting good results in the most common use cases.

AB-informatica-service, 2023-12-11:
Hi, any news here? I'm intresting to this argument

theGidy, 2024-02-18:
Its there any news?