arduino/ArduinoCore-mbed #631: Can't find Version 1.3.1 for Nano 33 BLE on Board Manager

calidiaz0596, 2023-02-25:
I am about to teach a class to several students and am following the guide that was given to me. In the guide, it instructs us to download the Nano 33 BLE Version 1.3.1 packages from the Board Manager, however this is not an option. The lowest the version goes is 2.0
[image]

Is there another way to download these packages? Do I need to have this specific version? 
Here is the information to the guide. https://drive.google.com/file/d/1tAU3Xb7oFZfufMTLEGkXDoXodJ_-VRs_/view

The class starts on Tuesday, so any help is greatly appreciated! 

If this is not the place for this question, please refer me a more appropriate place

hpssjellis, 2023-02-26:
@calidiaz0596 as a 30 year coding teacher and 7 year Robotics teacher, lessons being deprecated is our biggest pain.  I have spent 100's of hours work to fix minor changes in code for multiple courses.

Just try the newest version and see if it works. Then try the oldest version and see if that works. Really old versions can probably be found, but might produce other errors that are not worth it in the long run.  Can you give the link for the actual course?  It looks like it might be from this repository  https://github.com/tinyMLx/courseware/tree/master/edX/readings and if it is I know the Brian mentioned and can ask specific questions for you.

Typically a breaking change in the code causes one little part of the program not to work, terrible for beginners but typically easy for an Arduino specialist to fix. Give more information and someone here could probably help with it, although specific course issues are not Arduino's issue.

facchinm, 2023-02-27:
Hi @calidiaz0596 ,
you can still find core 1.3.1 under [DEPRECATED] label
[image]

Anyway, I'd follow @hpssjellis advice and try the latest core first. We strive to maintain complete compatibility between versions so updates should be as painless as possible.
Of course, if the library is targeting that particular version of the core, there's not a lot we can do, and you can go ahead with the old one.

closed 2023-02-27 by facchinm