arduino/ArduinoCore-mbed #632: Arduino MBED Version 4.X bootloader Question.

hpssjellis, 2023-03-03:
How important is it to load the MBED new bootloader for the Portenta? Mine is presently at Bootloader version: 23

What is the default question and answers that you would expect for developers? I get that final products would use the secure boot. 

1.  "Do you want to install/update the default Arduino bootloader? Y/[n]
Choosing "No", will install/update the MCUboot bootloader."  ?

Too scared to go farther. Looks like either choice does something to the board.

Is there a process to reverse the secure boot if you mess-up (I haven't yet, just asking). I assume there is no way to reverse a bad secure boot which would basically brick the board.

pennam, 2023-03-06:
Hi @hpssjellis,
The bootloader update `v24` introduced with mbed core release `4.0.2` is only for the Nicla Vision. The bootloader v24 for the portenta was introduced with release `3.4.1`.

Difference between version `v23` and `v24` is the way bootloader handles RTC clock at boot in order to read RTC backup registers: with newer cores > `3.4.1` and old bootloaders < `v24` there are some possible conditions in which backup registers are not retained with a board reset and OTA will not work.

For Portenta H7 the [STM32H747_manageBootloader](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino) sketch offers the user the possibility to flash [mcuboot-arduino-stm32h7](https://github.com/arduino/mcuboot-arduino-stm32h7) bootloader that offers the possibility to enable and evaluate some security features included in the [mcuboot](https://github.com/mcu-tools/mcuboot) bootloader.

The `mcuboot-arduino-stm32h7` can be used in two modes:
* Flashing only the bootloader but not the [keys](https://github.com/arduino/ArduinoCore-mbed/blob/64d42ffb0021352dd4a8acd53b461d8466523413/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino#L157) will make the bootloader work in the same exact way the "standard" bootloader is working, no changes and no security features enabled. 

* Flashing also the keys will enable the mcuboot security features and all new sketches have to be compliant to the mcuboot binary structure. The ide offers the possibility to build mcuboot sketches selecting Signature+Encryption in the Tools -> Security Setting manu of the IDE or using the [cli](https://arduino.github.io/arduino-cli/0.31/guides/secure-boot/) Building  the sketch in this way will encrypt the binary and add a signature that the bootloader will check before booting the sketch. The [STM32H747_manageBootloader](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino) sketch comes with a default set of keys that can be used only to evaluate the sketch creation and boot process.

Reverting from `mcuboot-arduino-stm32h7` is possible, but the route can be different depending how much things are "messed up":

* Option 1 - You have just flashed `mcuboot-arduino-stm32h7` but not the keys:
To revert to the "standard" bootloader you just need to reload the [STM32H747_manageBootloader](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino) sketch and answer Yes to `"Do you want to install/update the default Arduino bootloader? Y/[n]"`

* Option 2 - You have flashed `mcuboot-arduino-stm32h7` and keys:
To revert to the "standard" bootloader you need to build the [STM32H747_manageBootloader](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino) sketch with security settings  `Signature+Encryption`. Once the sketch is loaded answer Yes to  `"Do you want to install/update the default Arduino bootloader? Y/[n]"` In this way the keys will be erased and the "standard" bootloader will be flashed

* Option 3 - You have flashed `mcuboot-arduino-stm32h7` and keys but you have lost keys:
I  this situation you are not able to run any sketch on the board. The only way to restore it is to use an external programmer.

hpssjellis, 2023-03-06:
Thanks @pennam so unless I am doing OTA I should be fine with version Portenta V23. Obviously if I run into problems I will flash the new version v24 and see if it clears them up. I think I can close this issue.

closed 2023-03-06 by hpssjellis