arduino/ArduinoCore-mbed #644: Minimal example of peripheral SPI for Portenta H7

Andres9713, 2023-03-17:
Hello, 
I am trying to set up SPI communication between two Portenta H7's in order to facilitate _fast_ data transfer between them. The code for setting up the Portenta in controller ('master') mode is easy to follow and works fine (https://github.com/arduino/ArduinoCore-mbed/issues/131) - however, I have not been able to find any examples (minimal or otherwise) on how to setup the Portenta H7 as a _peripheral_ ('slave') SPI device. Does anyone know of such an example or resource?

alessandromrc, 2023-03-29:
Hi @Andres9713 if I'm not mistaken, the SPI library is implemented to only **send** data and not receive it

manchoz, 2023-03-30:
Hi @Andres9713,
You may want to experiment with the `SPISlave` class from the underlying MbedOS [SPISlave](https://os.mbed.com/docs/mbed-os/v6.16/apis/spislave.html). It might work if you don't initialize the default SPI bus.

Andres9713, 2023-03-30:
Thank you for your input! I might try it - _though_ since I couldn't figure out the SPI interface I ended up 'bit-banging' a solution with 14 parallel wires (8 data wires, 3 address wires (telling the receiver what part of the signal is currently being sent), 1 interrupt signal, and 2 bit sum wires (though I'm thinking these might not be that beneficial)). The sender sets its output pins to the correct values, sends an interrupt telling the receiver that data is ready and then waits 6 to 10 microseconds (I'm still working this number out) for the receiver to be done reading. I originally had the receiver send a 'data received' signal but this _really_ slowed things down. This homemade solution works well enough - I can send about 28k 32-bit float numbers per second this way - on the other hand it uses the majority of my useful pins. This is ok for now but is pretty limiting (since there are actually not that many useful pins left) and may become problematic later.

alessandromrc, 2023-03-30:
> Thank you for your inputI might try @manchoz suggestion - _though_ since I couldn't figure out the SPI interface I ended up 'bit-banging' a solution with 14 parallel wires (8 data wires, 3 address wires (telling the receiver what part of the signal is currently being sent), 1 interrupt signal, and 2 bit sum wires (though I'm thinking these might not be that beneficial)). The sender sets its output pins to the correct values, sends an interrupt telling the receiver that data is ready and then waits 6 to 10 microseconds (I'm still working this number out) for the receiver to be done reading. I originally had the receiver send a 'data received' signal but this _really_ slowed things down. This homemade solution works well enough - I can send about 28k 32-bit float numbers per second this way - on the other hand it uses the majority of my useful pins. This is ok for now but is pretty limiting (since there are actually not that many useful pins left) and may become problematic later.

This seems a really interesting way of dealing with it but surely GPIO consuming... keep us updated! 😊