arduino/ArduinoCore-mbed #650: Cannot Reinstall WiFi Firmware on Arduino Giga R1

ahmedalkabir, 2023-03-24:
for some reason, I couldn't apple to run WiFi as it stated that there's no firmware in the QSPI Flash, so I followed the sketch for WiFi Update but got stuck, so I tried to format the QSPI using "QSPIFormat Sketch"  and find out it has issue with formatting 

```
Available partition schemes:

Partition scheme 1
Partition 1: WiFi firmware and certificates 1MB
Partition 2: OTA and user data 13MB

Partition scheme 2
Partition 1: WiFi firmware and certificates 1MB
Partition 2: OTA 5MB
Partition 3: User data 8MB

Do you want to use partition scheme 1? Y/[n]
If No, partition scheme 2 will be used.

WARNING! Running the sketch all the content of the QSPI flash will be erased.
Do you want to proceed? Y/[n]
Error formatting WiFi partition

```

ahmedalkabir, 2023-03-24:
Still can't figure out the error and what causes it When I use WiFiFirmwareUpdater

```
No filesystem containing the WiFi firmware was found.
Usually that means that the WiFi firmware has not been installed yet or was overwritten with another firmware.

Formatting the filsystem to install the firmware and certificates...

Error Format Flash: -3101
Flashing /wlan/4343WA1.BIN file
Flashed 0%
```

josefwegner, 2023-03-24:
This seems to an issue with (some?) Giga boards. I already reported the same error to Arduino support, and the engineering team is looking into it. I will update the thread once I hear something from Arduino support.

ahmedalkabir, 2023-03-24:
@josefwegner I'm still working on the issue and it seems it's related to QSPIFBlockDevice As I can't read the content of the flash well, 

I followed the example of Mbed Api but the output for me doesn't make sense 

```
bd.init -> 0
--- Block device geometry ---
read_size:    1 B
program_size: 1 B
erase_size:   4096 B
size:         16777216 B
---
bd.read(0x2400ce00, 0, 15)
bd.read -> 0
--- Stored data ---
c8 ed ec ec ef a8 db fc ef fa e9 ef ed a9 88     ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
---
bd.erase(0, 4096)
bd.erase -> 0
bd.program(0x2400ce00, 0, 15)
bd.program -> 0
bd.read(0x2400ce00, 0, 15)
bd.read -> 0
--- Stored data ---
c8 ed ec ec ef a8 db fc ef fa e9 ef ed a9 88     ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
---
bd.deinit()
bd.deinit -> 0
--- done! ---

```

ahmedalkabir, 2023-03-24:
@facchinm, I was trying to figure out the issue of creating partition to establish a WiFi firmware, I found there's something wrong with QSPIFBlockDevice as Can't read the written contents, and I created a modified read function to see if there's problem with the flash or not, but it seems it's related to software issue as it can read contents by using a normal read instruction 0x03, here's testing code and output 

```
#include "mbed.h"
#include "QSPIFBlockDevice.h"
#include "drivers/QSPI.h"

QSPIFBlockDevice bd(QSPI_SO0, QSPI_SO1, QSPI_SO2, QSPI_SO3, QSPI_SCK, QSPI_CS, QSPIF_POLARITY_MODE_0, 40000000);

REDIRECT_STDOUT_TO(Serial);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial)
    ;
  // Initialize the block device
  int err = bd.init();

  int erase_val = bd.get_erase_value();
  Serial.println(bd.get_instrution());
  // Get device geometry
  mbed::bd_size_t read_size = bd.get_read_size();
  mbed::bd_size_t program_size = bd.get_program_size();
  mbed::bd_size_t erase_size = bd.get_erase_size();
  mbed::bd_size_t size = bd.size();

  printf("--- Block device geometry ---\n");
  printf("read_size:    %lld B\n", read_size);
  printf("program_size: %lld B\n", program_size);
  printf("erase_size:   %lld B\n", erase_size);
  printf("size:         %lld B\n", size);
  printf("---\n");

  char data[] = "QSPI Flash";
  size_t buffer_size = strlen(data) + program_size + 1;
  buffer_size = buffer_size - (buffer_size % program_size);
  char *buffer = new char[buffer_size];
  err = bd.erase(0, erase_size);
  strncpy(buffer, data, buffer_size);
  err = bd.program(buffer, 0, buffer_size);

  err = bd.read(buffer, 0, buffer_size);
  printf("using original call = bd.read() status %d -> %s\n", err, buffer);
  err = bd.read_2(buffer, 0, buffer_size);
  printf("using a modified which use A normal read by calling 0x03 instruction = bd.read_2() status %d -> %s\n", err, buffer);

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

output 

```
bd.init -> 0
--- Block device geometry ---
read_size:    1 B
program_size: 1 B
erase_size:   4096 B
size:         16777216 B
---
using  = bd.read() status 0 -> ⸮⸮⸮ɨ⸮⸮⸮⸮興⸮
using = bd.read_2() status 0 -> QSPI Flash
```

ahmedalkabir, 2023-03-31:
I created a quick handy solution for this issue check my [GigaWiFiSolver](https://github.com/ahmedalkabir/GigaWiFiSolver)
it might help you, @josefwegner 

@facchinm check it, maybe the team can resolve the issue quickly

josefwegner, 2023-04-20:
@ahmedalkabir: Thank you for your solution. I was on vacation, but I will check it will also solve my issue.

facchinm, 2023-07-13:
@josefwegner @ahmedalkabir sorry guys for taking so much time to fix this but we needed some brand new boards and it turned out being very difficult :slightly_smiling_face: 
I posted 30eacd2 that **should** solve the issue for good; the cause is still unknown but at least we know that it used to be ok due to [gpio_reset_at_init](https://github.com/arduino/mbed-os/blob/extrapatches-6.17.0/targets/targets.json#L4099C21-L4099C21) property being set to `true` during Giga development, while being removed on production. For some reasons, since QSPI initializes the pins inside its constructor, the status of pins physically close to IO2/IO3 (needed for Quad mode) was breaking the overall operation.
You can test a staging version of the core 4.0.4 (that will contain the fix) by adding https://downloads.arduino.cc/packages/package_giga_404_index.json to your Additional Sources.

Please let me know if it works so we can publish the official release :slightly_smiling_face:

josefwegner, 2023-07-13:
@facchinm:

Thank you for looking into this issue.

I am honest, I tried using the lastest git version yesterday. I managed to run QSPIFormat successfully once.

Instead of `Error formatting WiFi partition` I got `It's now safe to reboot or disconnect your board.`.

Afterwards I tried to run WiFiFirmwareUpdater, but I got the dreaded four long and four short red led flashes.

Flashing QSPIFormat again now also results in errors again.

It seems I broke the board for good this time.

facchinm, 2023-07-14:
@josefwegner if you didn't recompile the mbed library after cloning the core the patch would not be applied 🙂 try giving 4.0.4 a chance and I'm almost sure we'll be able to recover the board 🙏

josefwegner, 2023-07-14:
@facchinm:

Of course, I tried your version yesterday, too. BTW, the version linked in your file is *4.0.3*.

Sadly, it still does not help. Please find my details/steps below.

*~/.arduino15/arduino-cli.yaml*
```
board_manager:
  additional_urls: [
    https://downloads.arduino.cc/packages/package_giga_404_index.json,
    https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json,
    http://drazzy.com/package_drazzy.com_index.json,
    https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json,
    https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json,
    https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json,
    https://raw.githubusercontent.com/ehajo/arduino-boards/master/package_ehajo_index.json,
    https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
  ]
daemon:
  port: "50051"
directories:
  data: /home/pi/.arduino15
  downloads: /home/pi/.arduino15/staging
  user: /home/pi/Projects/arduino
library:
  enable_unsafe_install: true
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
sketch:
  always_export_binaries: false
```

Core is installed in `~/.arduino15/packages/arduino/hardware/mbed_giga/4.0.3`

```
$ arduino-cli core list
ID                  Installiert Neueste Name
arduino:avr         1.8.6       1.8.6   Arduino AVR Boards
arduino:mbed_giga   4.0.3       4.0.3   Arduino Mbed OS Giga Boards
arduino:mbed_nano   4.0.2       4.0.2   Arduino Mbed OS Nano Boards
arduino:mbed_rp2040 4.0.2       4.0.2   Arduino Mbed OS RP2040 Boards
arduino:megaavr     1.8.8       1.8.8   Arduino megaAVR Boards
arduino:sam         1.6.12      1.6.12  Arduino SAM Boards (32-bits ARM Cortex-M3)
arduino:samd        1.8.13      1.8.13  Arduino SAMD Boards (32-bits ARM Cortex-M0+)
ATTinyCore:avr      1.5.2       1.5.2   ATTinyCore
eHaJo:avr           2.2.0       2.2.0   eHaJo AVR Boards
MicroCore:avr       2.3.0       2.3.0   MicroCore
MightyCore:avr      2.2.2       2.2.2   MightyCore
MiniCore:avr        2.2.2       2.2.2   MiniCore
sandeepmistry:nRF5  0.7.0       0.7.0   Nordic Semiconductor nRF5 Boards
```

Copied QSPIFormat example to sketchpad, attached board, compiled, uploaded and ran it:
```
cp -r ~/.arduino15/packages/arduino/hardware/mbed_giga/4.0.3/libraries/STM32H747_System/examples/QSPIFormat ./
cd QSPIFormat
arduino-cli board attach -p /dev/ttyACM0 -b arduino:mbed_giga:giga
arduino-cli compile -u
```

Result:
```
Der Sketch verwendet 163436 Bytes (8%) des Programmspeicherplatzes. Das Maximum sind 1966080 Bytes.
Globale Variablen verwenden 48752 Bytes (9%) des dynamischen Speichers, 474872 Bytes für lokale Variablen verbleiben. Das Maximum sind 523624 Bytes.
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 2341:0366
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash  2MB   "
Downloading element to address = 0x08040000, size = 165420
Erase           [=========================] 100%       165420 bytes
Erase    done.
Download        [=========================] 100%       165420 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

Used platform     Version Pfad
arduino:mbed_giga 4.0.3   /home/pi/.arduino15/packages/arduino/hardware/mbed_giga/4.0.3
```

Connected with minicom:
```
minicom -c on -b 115200 -D /dev/ttyACM0
Available partition schemes:

Partition scheme 1
Partition 1: WiFi firmware and certificates 1MB
Partition 2: OTA and user data 13MB

Partition scheme 2
Partition 1: WiFi firmware and certificates 1MB
Partition 2: OTA 5MB
Partition 3: User data 8MB

Do you want to use partition scheme 1? Y/[n]
If No, partition scheme 2 will be used.
n
WARNING! Running the sketch all the content of the QSPI flash will be erased.
Do you want to proceed? Y/[n]
yError formatting WiFi partition
```

facchinm, 2023-07-17:
@josefwegner I'm super sorry for the mistake :slightly_frowning_face: The json now also contains the 4.0.4 package that will finally contain the fix to be tested. Sorry again

josefwegner, 2023-07-17:
@facchinm Thank you for fixing the json file. The version 4.0.4 indeed fixed the Flash issue. I could partition the flash and upload the WiFi firmware without any issue. A quick test to connect to my WiFi also worked :thumbsup:

Two other things I noticed that are not connected to this issue:
- What is correct order? First QSPIFormat and then WiFiFirmwareUpdater? Or the other way around?
- And the WiFi RTC example on https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-wifi is missing. It is the same as the WPA example.
Thanks again!

facchinm, 2023-07-17:
The order shouldn't matter if you want to use the 2 partitions setup, otherwise for the 3 partitions setup QSPIFormat must be run before WiFiFirmwareUpdater.
About the example, it looks like it's wrongly named and is just a variation on other basic examples. I'm pinging @arduino/ft_content to take a look :slightly_smiling_face: 
Apart from this, I'm super happy that it worked! Making its way to become official very soon :wink:

facchinm, 2023-07-25:
4.0.4 officially released, containing [30eacd2](https://github.com/arduino/ArduinoCore-mbed/commit/30eacd2dd7b983872087905084127f02f0e09352) :rocket: 
Thanks for your patience

closed 2023-07-25 by facchinm