arduino/ArduinoCore-mbed #651: Arduino nano 33 ble cost a lot of energy when powering on

haoliu0027, 2023-03-25:
When I just power the arduino nano 33 ble and it runs under the mode Board: "Arduino Mbed os nano board ->  Arduino nano 33 BLE" with empty loop code, it cost a lot of energy at the begining. It seems like that MBED initialize something, anyone know how to optimize this part of energy cost?

[image]

hpssjellis, 2023-03-25:
Have you seen these shots of thebportenta and nicla vision. If your only getting a 3mA power surge your doing great.

https://github.com/arduino/ArduinoCore-mbed/issues/619#issuecomment-1419789192

Makes me wondee if I have connected the nordic ppk2 properly. I will have to try my Nano33BleSense now.

hpssjellis, 2023-03-26:
@haoliu0027 I tried the Nan33BleSense with the Nordic PPK2 and 5V to VIN and got this from startup with a blank sketch, but I had not enabled power-output like your screen shot.

[image]

But when I load a blink with serial monitor I get a much different reading, more inline with my other MCU's  
[image]

If after I start the ppk2 I enable power output I then get a clean restart and get this graph, which is showing much more power usage than your graph. For my graph the Nano33BleSense power LED is on. 

[image]