arduino/ArduinoCore-mbed #662: Mismatch between CameraCaptureRawBytes and CameraRawBytesVisualizer

aliphys, 2023-04-13:
# Problem
:disappointed: This is what happens when `useGreyScale` is set to `true` for the Nicla Vision:
[image]
:smile: This is what is observed, when `userGreyScale` is set to `false` for the Nicla Vision
[image]

This creates three problems:
- The header comment and code in CameraRawBytesVisualizer conflict in what the default colour mode is
- From a UX perspective (cc @sebromero ) the User can face additional hurdles in testing out a basic function (e.g [this forum post](https://forum.arduino.cc/t/giga-r1-camera-example-problem/1112691/2))
- There is no clear pathway for users to realise the problem, upon seeing the corrupt image.

# Details
The [CameraCaptureRawBytes](https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/libraries/Camera/examples/CameraCaptureRawBytes/CameraCaptureRawBytes.ino) sketch is used to send a video stream over Serial to the computer. Which can then be displayed with the [CameraRawBytesVisualizer](https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/libraries/Camera/extras/CameraRawBytesVisualizer/CameraRawBytesVisualizer.pde#L19) Processing sketch.

A series of pre-directives, configures the `IMAGE_MODE` to be either `CAMERA_RGB` (Nicla Vision and GIGA boards) or `CAMERA-GREYSCALE` (Portenta H7 family). 
https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/libraries/Camera/examples/CameraCaptureRawBytes/CameraCaptureRawBytes.ino#L3-L19

In the default configuration, the accompanying CameraRawBytesVisualizer sketch is configured to be compatible with `CAMERA_GREYSCLE`, even though the header comment states that the Processing sketch is for RGB.
https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/libraries/Camera/extras/CameraRawBytesVisualizer/CameraRawBytesVisualizer.pde#L1-L7
https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/libraries/Camera/extras/CameraRawBytesVisualizer/CameraRawBytesVisualizer.pde#L19-L20

--- 
Thanks @marqdevx , @Hannes7eicher and @jacobhylen for letting me pick your brains :brain:

aliphys, 2023-04-13:
Replicated issue on Arduino GIGA R1 WiFi with OV7675 cc @karlsoderby

`useGrayScale` set to `false`
[image]

`useGreyScale` set to `true`
[image]

aliphys, 2023-04-27:
Possible solution: Use [keyPressed()](https://processing.org/reference/keyPressed_.html) method in Processing to switch the boolean value for `useGrayScale`.  

Something like:
```
void keyPressed() {
  if (key == ' ') {
    useGrayScale = !useGrayScale;
  }
}
``` 

This does not require the uses to change the `useGrayScale` variable manually, and can be switched without restarting the Processing sketch.

rneurink, 2023-06-12:
See `case CAMERA_GRAYSCALE:`, it is commented out and currently GRAYSCALE sets to camera to use a bayer output. This is why the grayscale image looks really weird. If you uncomment these lines you can get it to work. 
https://github.com/arduino/ArduinoCore-mbed/blob/9795c3c22e009796d3f1a283fc6aa1b3a8575e43/libraries/GC2145/gc2145.cpp#L892-L926