arduino/ArduinoCore-mbed #677: Usage of I2C and Serial port

lu7did, 2023-05-02:
Hi,

Trying to port a firmware written for the Arduino core by Earle Philtower into mBed.

I do need to use a serial port at UART0 pins 16,16 (GPIO12/13) and a I2C device connected with pin 21,22 (GPIO 16/17), 
currently I'm defining the first with 

For the I2C

Wire.setSDA(16);
Wire.setSCL(17);
Wire.begin();

For the serial

 Serial1.setTX(12);
 Serial1.setRX(13);
 Serial1.begin(115200);

Would ask for a hint on how to do this in the mbed core.

Is there a documentation on the core somewhere?

Thanks in advance, Pedro.

facchinm, 2023-05-03:
Hi Pedro,
in this core you can get the same result by using
```

closed 2023-05-03 by facchinm

lu7did, 2023-05-03:
Thank you very much, appreciate!