arduino/ArduinoCore-mbed #704: openocd 0.11.0-arduino2 does not know target/rp2040.cfg

josefwegner, 2023-07-23:
Hi,

Set up a spare Raspberry PI Pico as a Picoprobe (as described in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf Chapter A) and tried to upload a sketch using a programmer instead of using picotool:

```
arduino-cli board attach -p /dev/ttyACM0 -b arduino:mbed_rp2040:pico 
arduino-cli compile
arduino-cli upload -P cmsis-dap -v
```

Sadly, it fails as openocd is trying an uknown config:
```
"/home/pi/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/bin/openocd" -d2 -s "/home/pi/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/share/openocd/scripts/" -f interface/cmsis-dap.cfg  -f target/rp2040.cfg -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; program {/tmp/arduino/sketches/CF5F9525C622B4069B93C96C38646A84/neopixel.ino.elf}; reset run; shutdown"
Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:55)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

embedded:startup.tcl:26: Error: Can't find target/rp2040.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 26
Fehlgeschlagene Programmierung: Hochladefehler: exit status 1
```

boards.txt is using a wrong target:
```
pico.bootloader.tool=openocd
pico.bootloader.tool.default=openocd
pico.bootloader.config=-f target/rp2040.cfg
pico.bootloader.programmer=-f interface/cmsis-dap.cfg
```

The bundled openocd-0.11.0-arduino2 does only have the following target:
`~/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/share/openocd/scripts/target/rp2040-core0.cfg`

Please either rename the file to rp2040.cfg, or update boards.txt to point to the correct file.

josefwegner, 2023-07-23:
Using rp2040-core0.cfg also does not seem to work for me:
```
"/home/pi/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/bin/openocd" -d2 -s "/home/pi/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/share/openocd/scripts/" -f interfac
e/cmsis-dap.cfg  -f target/rp2040-core0.cfg -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; program {/tmp/arduino/sketches/CF5F9525C622B4069B93C96C38646A84/neopixel.ino.elf}; reset run; shutdown"
Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:55)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

cortex_m reset_config sysresetreq

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6609CB2D31C8A28
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Error: CMSIS-DAP command CMD_DAP_SWJ_CLOCK failed.
```

But using a stock openocd-0.12 would work fine with the given command line and target rp2040.cfg:
```
openocd -d2 -s "/home/pi/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2/share/openocd/scripts/" -f interface/cmsis-dap.cfg  -f target/rp2040.cfg -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; program {/tmp/arduino/sketches/CF5F9525C622B4069B93C96C38646A84/neopixel.ino.elf}; reset run; shutdown"
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6609CB2D31C8A28
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : Padding image section 1 at 0x100174c4 with 60 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10017500 .. 0x1001ffff
** Programming Finished **
shutdown command invoked
```

wzqvip, 2023-12-09:
met the same problem. 

Using Arduino IDE 2.2.1 with RP2040 on windows.

Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0

embedded:startup.tcl:26: Error: Can't find target/rp2040.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Failed to burn bootloader: uploading error: exit status 1

wawanUnic, 2024-09-01:
Me too:

Open On-Chip Debugger 0.11.0+dev-gab95bac57-dirty (2021-05-11-10:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

embedded:startup.tcl:26: Error: Can't find target/rp2040.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Failed programming: uploading error: exit status 1