arduino/ArduinoCore-mbed #705: USBMouseKeyboard

dmcke5, 2023-07-26:
I'm unable to get keycodes to produce actual button presses, they just seem to input characters instead.

For example when I call: Keyboard.key_code(0x26); this should represent the up arrow being pressed on the keyboard, but instead I get the number 7 when I test it.
The same thing happens for all the Arrow keys:
0x25 = 5
0x26 = 7
0x27 = '
0x28 = 9

The codes for Enter and Escape (0x0D and 0x1B) don't seem to do anything at all.

These are the only buttons I've tested so far, so I'm unsure if any of them work.
I'm on version 4.04 of the library which I believe is the latest.

Any help would be appreciated!

facchinm, 2023-07-26:
Hi @dmcke5 ,
to send keycodes directly you should use `Keyboard.key_code_raw()` API :wink: Let me know if it works!

dmcke5, 2023-07-30:
> Hi @dmcke5 , to send keycodes directly you should use `Keyboard.key_code_raw()` API 😉 Let me know if it works!

Can confirm, I copied key_code_raw() from the Keyboard library over to the USBMouseKeyboard library on my end and it works fine.