arduino/ArduinoCore-mbed #707: WebSerial to install compiled files

hpssjellis, 2023-07-27:
I am doing some machine learning web browser work with WebSerial and even sites like p5.js have webSerial working for Arduino's  see their online viewer [here](https://editor.p5js.org/gohai/sketches/X0XD9xvIR). Does anyone know  if a simple webSerial interface can be used to install a pre-compiled Arduino binary? Perhaps @facchinm or @sebromero

I am guessing it can't, but thought I would ask. WebSerial works when the Arduino is in normal and DFU mode so I am curious if we can just write the binary file to the Arduino?

By the way, my very cool webSerial page is called [tinyMLjs](https://hpssjellis.github.io/tinyMLjs/public/acceleration/a00-best-acceleration.html)  and works really well with the all Arduinos but specifically the Nano33BleSense

facchinm, 2023-07-27:
Hi Jeremy,
for dfu upload via WebUSB (not WebSerial) you can use this project https://devanlai.github.io/webdfu/dfu-util/ 
I just tested with a Portenta H7 and it works great as long as you specify Internal Flash interface and 0x8040000 as starting offset.

hpssjellis, 2023-07-27:
Very interesting @facchinm I will look into that. 

By the way, at a machine learning conference in Trieste Italy I met some Arduino Employees: Jose, Yu and David Cuartielles. I hope one of them mentioned to you how much I appreciate your coding help. You really have helped with a lot of the open source work I do.

closed 2023-07-27 by hpssjellis

hpssjellis, 2023-07-28:
Wow @facchinm I can't believe  how well it works on both chrome and edge.  Download and upload firmware.bin easily from the Portenta  using vendor 0x2341.  The github is at https://github.com/devanlai/webdfu/ and I can't believe how old it is some of the files are 6 years old. I will try to connect with him and see if the development is still active.   

[image]

[image]

hpssjellis, 2023-07-29:
Wow this code almost worked n my Android phone connected to the portenta. It fully downloaded the firmware.bin file but had an issue uploading the firmware.bin file.  Note sure if the downloaded file was corrupted when I tried to upload it. I should try a properly made firmware.bin file.  Very cool if that could actually work.