arduino/ArduinoCore-mbed #723: ArduinoBLE on Giga R1 can only run on the main core

Gerriko, 2023-09-01:
I tried helping someone get ArduinoBLE to work on the M4 core but BLE.begin() crashes - it does not return 0 or 1.
I believe this is because of the restriction that Serial can only run on the main core and LBEE5KL1DX-883 (U5) uses UART7 (HCI).
See: https://forum.arduino.cc/t/how-to-use-bluetooth-in-co-processors-sketch/1163997/3

Is there any way to create an exception for UART7, or is there any way one can use software serial in this case.

Gerriko, 2023-09-01:
Apologies - this issue is probably more related to the ArduinoBLE library.
Anyway, on a somewhat related topic about handling conditions in different cores on the Giga R1, I have noticed that with the Giga there is only one precompiler define (ARDUINO_GIGA), whereas for the Portenta, you have for #if define "ARDUINO_PORTENTA_H7_M7" and "ARDUINO_PORTENTA_H7_M4". Maybe 2 options are needed for the Giga too.

Nocentinia, 2023-10-23:
However, it is absurd how the Arduino staff abandons us when these problems arise.
I have the same problem, when I use BLE with any other library the chip goes crazy and freezes.
then I wanted to try using the other core to manage the connection but it still doesn't work!!
and no arduino staff trying to fix things, it's embarrassing

alranel, 2023-11-07:
I'm closing this one since it's being handled in the proper repository (https://github.com/arduino-libraries/ArduinoBLE/issues/330). As of today, a pull request is available containing the fix.

closed 2023-11-07 by alranel