arduino/ArduinoCore-mbed #724: Compile error in v4.0.4

zfields, 2023-09-02:
```
~/.arduino15/packages/arduino/hardware/mbed_nano/4.0.4/cores/arduino/api/CanMsg.h:71:71: error: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32_t {aka long unsigned int}' [-Werror=format=]
     len = snprintf(buf, sizeof(buf), "[%08X] (%d) : ", id, data_length);
```

I was going to offer a patch, but I cannot find the path.

per1234, 2023-09-03:
Hi @zfields. The files under the `cores/arduino/api` folder contain architecture independent code that can be shared by the cores of all Arduino boards platforms, so they are maintained in a dedicated repository, from which they are pulled while preparing the release packages of this platform that are distributed via Boards Manager:

https://github.com/arduino/ArduinoCore-API/blob/master/api/CanMsg.h

A proposed fix has already been submitted here: https://github.com/arduino/ArduinoCore-API/pull/198. You are welcome to review that PR.

zfields, 2023-09-06:
I see it was merged. What steps will need to transpire, before it becomes available in the CLI?

facchinm, 2023-09-06:
4.0.6 is in the oven, it will be published before the end of the week :wink:

closed 2023-09-26 by facchinm