arduino/ArduinoCore-mbed #737: GIGA R1 - SerialX.flush() does not work

KurtE, 2023-10-17:
As I mentioned in the forum post:
[SerialX (X=1-4) are unbuffered? surprised ](https://forum.arduino.cc/t/serialx-x-1-4-are-unbuffered-surprised/1178363?u=kurte)

The call to SerialX.flush() does not work as I expected, nor does it comply with the documentation.
https://www.arduino.cc/reference/en/language/functions/communication/serial/flush/

>Waits for the transmission of outgoing serial data to complete. (Prior to Arduino 1.0, this instead removed any buffered incoming serial data.)

To reproduce:
```
void setup() {
  Serial1.begin(115200);
  pinMode(2, OUTPUT);
}

void loop() {
  digitalWrite(2, HIGH);
  Serial1.print("ABCDEFGHIJKLMNOPQRSTUVWXYZ");
  Serial1.flush();
  digitalWrite(2, LOW);
  delay(1);
  digitalWrite(2, HIGH);
  for (char ch = 'a'; ch <= 'z'; ch++) Serial1.write(ch);
  Serial1.flush();
  digitalWrite(2, LOW);

  delay(1000);
}

```
Results captured by Logic analyzer:
[image]

The hardware appears to have a Transfer Complete (TC) status, which looks like it should be able to do this properly.
Alternatively, is to add some random delay after your transfers to make up for this issue, which I personally would prefer to avoid.

Otherwise, you are likely to see variations of this same issue, like: #389

facchinm, 2023-10-17:
Hi @KurtE ,
we tried to get this fixed long time ago in mbed-os https://github.com/ARMmbed/mbed-os/pull/14600 but the proposal was then rejected. At the moment, there is no cross architecture way to make sure the FIFO is flushed, so we have to deal with it (until we find a better solution).

multiplemonomials, 2024-01-12:
@pennam If you submit https://github.com/ARMmbed/mbed-os/pull/14600 to mbed-ce/mbed-os, I will review and merge!