arduino/ArduinoCore-mbed #738: GIGA R1 - SPI - SPI_INTERFACES_COUNT is not defined

KurtE, 2023-10-18:
I am debugging why the Adafruit_ILI9341 library does not work on the GIGA board when you are trying to use the 
SPI pins that are on the outer pins (11-13), which are defined as SPI1.  

I debugged it up to Adafruit SPITFT code in GFX, that has code like:
```
    if (
#if !defined(SPI_INTERFACES_COUNT)
        1
#else
#if SPI_INTERFACES_COUNT > 0
        (hwspi._spi == &SPI)
#endif
#if SPI_INTERFACES_COUNT > 1
        || (hwspi._spi == &SPI1)
#endif
#if SPI_INTERFACES_COUNT > 2
        || (hwspi._spi == &SPI2)
#endif
#if SPI_INTERFACES_COUNT > 3
        || (hwspi._spi == &SPI3)
#endif
#if SPI_INTERFACES_COUNT > 4
        || (hwspi._spi == &SPI4)
#endif
#if SPI_INTERFACES_COUNT > 5
        || (hwspi._spi == &SPI5)
#endif
#endif // end SPI_INTERFACES_COUNT
    ) {
      Serial.println("\t *** Start the SPI object ***");
      hwspi._spi->begin();
    }
  } else if (connection == TFT_SOFT_SPI) {

```
Since SPI_INTERFACES_COUNT defined, it will not check against SPI1.  

In my local copy I have defined it which is till not properly working on the SPI1 pins.   It does work when you configure
your code by defining all of the pins, which then uses software SPI.  It also appears to work if I use the SPI pins on the six
pins connector.

facchinm, 2023-10-30:
@KurtE I'd close the issue since the problem lies in the library, or do you think that adding `SPI_INTERFACES_COUNT` in the core would be beneficial too? That define is not "standard" but it would be useful to have a standard one sooner or later.

KurtE, 2023-10-30:
@facchinm - this is up to you, but I know I have written code in the past that does similar.  And probably at least half of your boards do export this value.   

Without it some libraries may not build whereas others, may simply then assume that the device only supports SPI object.  And in the case of GIGA that could hurt as the main SPI that many may use are on the ones on the shield pins (10-13) which is SPI1