arduino/ArduinoCore-mbed #742: OPTA TCP problems

sstaub, 2023-10-24:
I have massive problems using TCP on OPTA. After sending some packets (sometimes only 1) the OPTA stops working. It seems the program is blocking.
An interesting behavior is sending to the PacketSender (https://packetsender.com) it works but not with other applications which my software is written for (professional lighting applications). I'm sending only very small packets (max. 60 bytes).
There is also no way to set the TCP socket to nonblocking (maybe over mbed API?), also the functions Ethernet.setRetransmissionTimeout() and Ethernet.setRetransmissionCount() are not available.
On mbed site I see that there is an issue https://github.com/ARMmbed/mbed-os/issues/15429 for the used microcontroller.

mbedOS Opta v4.0.8
IDE 2.2.1 macOS Sonoma

sstaub, 2023-10-28:
I found this on mbed forum https://forums.mbed.com/t/sending-packets-with-tcp-sockets-continuously/15237
On this way the TCP functionality is useless.

sstaub, 2023-10-28:
Here is the example code, pressing a button send a message. You can do it only once, then no more.
```cpp
#include "Ethernet.h"

#define OSC_MESSAGE_SIZE 32
#define DEBOUNCE_DELAY_MS 50

uint8_t buttonState;
uint8_t lastButtonState;
uint32_t lastDebounceTime;

IPAddress optaIP(10, 101, 1, 201);
uint16_t optaPort = 9000;
IPAddress receiverIP(10, 101, 1, 100);
uint16_t receiverPort = 9000;

// "/eos/ping\x00\x00\x00,\x00\x00\x00"
// 2F 65 6F 73 2F 70 69 6E 67 00 00 00 2C 00 00 00 //TCP 16 chars
// C0 2F 65 6F 73 2F 70 69 6E 67 00 00 00 2C 00 00 00 C0 //TCP Slip 18 chars
// 00 00 00 10 2F 65 6F 73 2F 70 69 6E 67 00 00 00 2C 00 00 00 //TCP Length 20 chars

uint8_t messageTCP[OSC_MESSAGE_SIZE] = {0x2F, 0x65, 0x6F, 0x73, 0x2F, 0x70, 0x69, 0x6E, 0x67, 0x00, 0x00, 0x00, 0x2C, 0x00, 0x00, 0x00};
uint8_t messageTCPSLIP[OSC_MESSAGE_SIZE] = {0xC0, 0x2F, 0x65, 0x6F, 0x73, 0x2F, 0x70, 0x69, 0x6E, 0x67, 0x00, 0x00, 0x00, 0x2C, 0x00, 0x00, 0x00, 0xC0};
uint8_t messageTCPLENGTH[OSC_MESSAGE_SIZE] = {0x00, 0x00, 0x00, 0x10, 0x2F, 0x65, 0x6F, 0x73, 0x2F, 0x70, 0x69, 0x6E, 0x67, 0x00, 0x00, 0x00, 0x2C, 0x00, 0x00, 0x00};;

EthernetClient tcp;

void setup() {
	pinMode(BTN_USER, INPUT);
	pinMode(LED_BUILTIN, OUTPUT);
	Ethernet.begin(optaIP);
	tcp.connect(receiverIP, receiverPort);
	}

void loop() {
	uint8_t reading = digitalRead(BTN_USER);
	if (reading != lastButtonState) {
		lastDebounceTime = millis();
		}
	if ((millis() - lastDebounceTime) > DEBOUNCE_DELAY_MS) {
		if (reading != buttonState) {
			buttonState = reading;
			if (buttonState == HIGH) {
				digitalWrite(LED_BUILTIN, LOW);
				}
			else { // USER_BTN pressed
				digitalWrite(LED_BUILTIN, HIGH);
				if (!tcp.connected()) {
					tcp.connect(receiverIP, receiverPort);
					}
				//tcp.write(messageTCP, 16);
				//tcp.write(messageTCPLENGTH, 20);
				tcp.write(messageTCPSLIP, 18);
				}
			}
		}
		lastButtonState = reading;
	}
```

facchinm, 2023-10-30:
@manchoz may you try to replicate the issue?