arduino/ArduinoCore-mbed #745: Pico change SPI flash clock divider

luki34567780, 2023-10-25:
Is it currently possible to change the SPI flash clock divider for the Raspberry Pi Pico? The way to do this with the Pico SDK would be to define `PICO_FLASH_SPI_CLKDIV`, but this doesn't seem to have any effect here.

sabas1080, 2023-10-27:
x2

luki34567780, 2023-11-21:
I've switched to [earlephilhower's core](https://github.com/earlephilhower/arduino-pico). With it you can just modify the board config to change the SPI flash divider (find your board in `C:\Users\<user>\.platformio\platforms\raspberrypi@src-ff76a3915224135aafad379817f41edd\boards` and change "boot2_source" from `<something>_2_padded_checksum.S` to `<something>_4_padded_checksum.S`. Also make sure that you got the correct board (there is a `pico.json` and a `rpipico.json`)