arduino/ArduinoCore-mbed #753: WiFi library - os crash on Giga R1 when WiFiClient global

megacct, 2023-11-07:
As per the thread below, os crashes on Giga R1 when WiFiClient is declared as global.

[https://forum.arduino.cc/t/call-to-server-available-fails-if-client-is-a-global-variable/1137604](url)

alranel, 2023-11-07:
@facchinm if I understand correctly the logic that this helper method implements, why don't we use shared_ptr instead?

megacct, 2023-11-13:
@facchinm @alranel guys, any update regarding this issue? This functionality is essential for me and a blocker to my GIGA migration. Thanks for all the work

facchinm, 2023-11-13:
We are working on it, don't worry

rickclemenzi, 2023-12-12:
This solved the identified problem, but there are more: status() is not returning WL_CONNECTED when it is connected.  It's returning '1' which per wl_definitions.h is WL_NO_SSID_AVAIL.  It did connect to a known SSID so this cannot be correct.

JAndrassy, 2023-12-12:
@rickclemenzi client.status doesn't return WL_ constants. it should return TCP states
https://github.com/JAndrassy/Arduino-Networking-API/blob/main/ArduinoNetAPIDev.md#status-optional

rickclemenzi, 2023-12-12:
@JAndrassy Thanks and I hear you, but there is a significant disconnect here that needs clearing up.  WiFi.status() per [the online manual](https://www.arduino.cc/reference/en/libraries/wifi/wifi.status/) is clearly WL_... based, and WiFi.cpp from the mbed_giga library is WL_.. based as well (my address C:\Users\RC2\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.0.10\libraries\WiFi).  It seems there are two half separate library threads involved, but I don't see how the WiFi.cpp is not the one setting status in for WiFiClient.  Sorry to be thick, but I don't see how all these facts fit together and would appreciate clarification.  The documentation was 100% correct for my earlier ports to the Uno Wifi Rev 2.

JAndrassy, 2023-12-12:
@rickclemenzi this issue is about the WiFiClient

closed 2024-10-13 by megacct