arduino/ArduinoCore-mbed #758: GIGA R1 - analogRead(A12) returns A8 value

megacct, 2023-11-12:
and analogRead(A13) returns A9 value. However, using Adafruit advancedAnalog library the correct values for A12 and A13 are returned

facchinm, 2023-11-13:
Mmmmh no, I retested just to make sure and it works as intended (reads A13/DAC1 value when `analogRead(A13)` is called).
Are you sure you are not using `digitalRead()` ? Because pins from A8 to A11 are analog only and get skipped in the count when using as digital https://github.com/arduino/ArduinoCore-mbed/blob/main/variants/GIGA/pins_arduino.h#L54-L55 since they can't be used as digital pins.

facchinm, 2023-11-13:
Got it, I've been able to replicate.
@Rocketct can you take a look?

Rocketct, 2023-11-13:
sure @facchinm

Rocketct, 2023-12-01:
https://github.com/arduino/ArduinoCore-mbed/pull/784 this pr should fix the issue

closed 2023-12-04 by facchinm

white-ar, 2024-07-25:
Hello, I've just come across the issue with a Giga R1.  Upgrading to V 4.1.5 of the board library _temporarily_ fixed the issue, however it has returned.  Here is the very simple sketch:

@facchinm @Rocketct is there a way of addressing A9 using MBED calls?  (as a work around).  Thank you. 

`
#include "Arduino.h"
#include "RPC.h"

#define SERVO_POT_1     A9

void setup() {
  // put your setup code here, to run once:
  RPC.begin();

  analogReadResolution(16); //GIGA has 16 bit ADC  (0-65535)  

}

void loop() {
  // put your main code here, to run repeatedly:

int temp = analogRead(SERVO_POT_1);

}`

Error message:
C:\...\AppData\Local\Temp\arduino\sketches\D2129B132D44D2F5F57EC30CA80C2E2B\sketch\sketch_jul25a.ino.cpp.o: In function `loop':
C:\...\AppData\Local\Temp\.arduinoIDE-unsaved2024625-43476-ja9cee.pq8d\sketch_jul25a/sketch_jul25a.ino:20: undefined reference to `A9'
C:\...\AppData\Local\Temp\.arduinoIDE-unsaved2024625-43476-ja9cee.pq8d\sketch_jul25a/sketch_jul25a.ino:20: undefined reference to `analogRead(PureAnalogPin)'
collect2.exe: error: ld returned 1 exit status

Using library RPC at version 1.0 in folder: C:\...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\RPC 
Using library rpclib at version 1.0.0 in folder: C:\...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\rpclib 
Using library openamp at version 1.0 in folder: C:\...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\openamp_arduino 
exit status 1

Compilation error: exit status 1

Any ideas?