arduino/ArduinoCore-mbed #780: Arduino Giga attachInterrupt() resets pinMode Pullup/Pulldown

N9W9, 2023-11-28:
When using the attachInterrupt() function set Pullups/Pulldowns are reset, the Pin is left floating.

When setting the pinMode after attaching the interrupt, it does work fine. However this is inconsistent with most examples which set the pinMode before attaching the Interrupt (e.g. https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/; also on the Arduino Giga Cheat Sheet)
It is also inconsistent with Arduino Mega, where the pinMode does not get reset after the attachInterrupt() call.

Example Code:
``` cpp
volatile int count = 0;
const byte pin = 2;

void setup() {
  Serial.begin(9600);
  //pinMode(pin, INPUT_PULLDOWN);    // Setting the Pin Mode here does not work!
  attachInterrupt(digitalPinToInterrupt(pin), ISR_count, RISING);
  pinMode(pin, INPUT_PULLDOWN);       // Pin Mode has to be set after the attachInterrupt()
  delay(100);
  Serial.println("Start");
}

void loop() {
  Serial.println(String("Pin State: ") + digitalRead(pin) + String(" Interrupt Count: ") + count);
}

void ISR_count() {
  count++;
}
```

rchesterr, 2024-02-19:
Hit the same issue, as above moving pinmode after attachinterrupt fixes this pullup issue, but means the activation of the pull-up (or down) can cause an interrupt.

facchinm, 2024-02-21:
This is being tackled by this PR https://github.com/arduino/ArduinoCore-mbed/pull/256 , but was never merged due to the inefficiency of the solution. Any idea is appreciated :slightly_smiling_face: