arduino/ArduinoCore-mbed #789: Arduino Giga R1 crashes with attach interrupt

Heinrich222, 2023-12-04:
This Testsketch will cause a "crash" with the Giga Board just after upload ( Red LED blinking ).
According to the Documentation it should work.

but: 
As soon as I remove the two lines marked with xyz the Sketch works fine.

crash occurs also:
- when using pin 52,53 instead of 40 / 42
- when using pin 40 or 42 exclusive
- seems independant from number of attached Interrupts/Pins

???

Thanks for any Hint.

`volatile int MM;
volatile int i;

void setup() {

  attachInterrupt(digitalPinToInterrupt(34), ISR_34, FALLING);
  attachInterrupt(digitalPinToInterrupt(36), ISR_36, FALLING);
  attachInterrupt(digitalPinToInterrupt(38), ISR_38, FALLING);
  attachInterrupt(digitalPinToInterrupt(40), ISR_40, FALLING); // xyz
  attachInterrupt(digitalPinToInterrupt(42), ISR_42, FALLING); // xyz
  attachInterrupt(digitalPinToInterrupt(44), ISR_44, FALLING);
  attachInterrupt(digitalPinToInterrupt(46), ISR_46, FALLING);
  attachInterrupt(digitalPinToInterrupt(48), ISR_48, FALLING);
  attachInterrupt(digitalPinToInterrupt(50), ISR_50, FALLING);
  attachInterrupt(digitalPinToInterrupt(22), ISR_22, FALLING);

  pinMode(34, INPUT_PULLUP);
  pinMode(36, INPUT_PULLUP);
  pinMode(38, INPUT_PULLUP);
  pinMode(40, INPUT_PULLUP);
  pinMode(42, INPUT_PULLUP);
  pinMode(44, INPUT_PULLUP);
  pinMode(46, INPUT_PULLUP);
  pinMode(48, INPUT_PULLUP);
  pinMode(50, INPUT_PULLUP);
  pinMode(22, INPUT_PULLUP);
  
  Serial.begin(9600);
}

void loop() {
  if (MM != 0) {
    i++;
    Serial.print(i);
    Serial.print(" -- >");
    Serial.println(MM);
    delay(100);
    MM = 0;
  }
}

void ISR_42() {
  MM = 42;
}
void ISR_40() {
  MM = 40;
}
void ISR_38() {
  MM = 38;
}
void ISR_36() {
  MM = 36;
}
void ISR_34() {
  MM = 34;
}
void ISR_44() {
  MM = 44;
}
void ISR_46() {
  MM = 46;
}
void ISR_48() {
  MM = 48;
}
void ISR_50() {
  MM = 50;
}
void ISR_22() {
  MM = 22;
}
`

facchinm, 2023-12-05:
Hi @Heinrich222 , the Giga allows a maximum of 16 interrupts at the same time, and the coexistence is based on STM32 peculiar way of handling them (like, PA_**0** will share the interrupt with PB_**0**, PC_**0** and so on). If you instantiate 2 interrupts on the same line the board will "crash" (report it as invaliud).
To associate pin name and pin number you can use this table https://github.com/arduino/ArduinoCore-mbed/blob/main/variants/GIGA/variant.cpp#L31-L160

closed 2023-12-05 by facchinm

Heinrich222, 2023-12-05:
Hello
Thank you very much.
My application works fine now.

best regards
Heinrich

CombiesGit, 2023-12-05:
My suggestion:
This could be mentioned in the documentation for the board.

> All GPIO pins on the GIGA R1 can be used for interrupts.

https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet#interrupts

facchinm, 2023-12-05:
>My suggestion:
>This could be mentioned in the documentation for the board.

Very nice suggestion! @karlsoderby :point_up: :slightly_smiling_face:

racingtogreen, 2024-03-19:
I note Pin22 included. I just ran a sketch using only 4 interrupts, 22-25, and got a consistent board crash. pins 24-27 were fine. Might be an issue with Pin22 ?

per1234, 2024-03-19:
Hi @racingtogreen 

>I just ran a sketch using only 4 interrupts, 22-25, and got a consistent board crash.

The reason was already explained:

https://github.com/arduino/ArduinoCore-mbed/issues/789#issuecomment-1840246454

> the coexistence is based on STM32 peculiar way of handling them (like, PA_0 will share the interrupt with PB_0, PC_0 and so on). If you instantiate 2 interrupts on the same line the board will "crash" (report it as invaliud).
To associate pin name and pin number you can use this table https://github.com/arduino/ArduinoCore-mbed/blob/main/variants/GIGA/variant.cpp#L31-L160

https://github.com/arduino/ArduinoCore-mbed/blob/ac7d14eb8404ad9c396eabc36a655c1183e8f3e4/variants/GIGA/variant.cpp#L57-L60

Note that pins 22 `PJ_12` and 24 (`PG_12`) use "line" 12.

> pins 24-27 were fine.

That is because each of these pins use a different "line":

https://github.com/arduino/ArduinoCore-mbed/blob/ac7d14eb8404ad9c396eabc36a655c1183e8f3e4/variants/GIGA/variant.cpp#L59-L62

racingtogreen, 2024-03-19:
> Hi @racingtogreen
> 
> > I just ran a sketch using only 4 interrupts, 22-25, and got a consistent board crash.
> 
> The reason was already explained:
> 
> [#789 (comment)](https://github.com/arduino/ArduinoCore-mbed/issues/789#issuecomment-1840246454)
> 
> > the coexistence is based on STM32 peculiar way of handling them (like, PA_0 will share the interrupt with PB_0, PC_0 and so on). If you instantiate 2 interrupts on the same line the board will "crash" (report it as invaliud).
> > To associate pin name and pin number you can use this table https://github.com/arduino/ArduinoCore-mbed/blob/main/variants/GIGA/variant.cpp#L31-L160
> 
> https://github.com/arduino/ArduinoCore-mbed/blob/ac7d14eb8404ad9c396eabc36a655c1183e8f3e4/variants/GIGA/variant.cpp#L57-L60
> 
> Note that pins 22 `PJ_12` and 24 (`PG_12`) use "line" 12.
> 
> > pins 24-27 were fine.
> 
> That is because each of these pins use a different "line":
> 
> https://github.com/arduino/ArduinoCore-mbed/blob/ac7d14eb8404ad9c396eabc36a655c1183e8f3e4/variants/GIGA/variant.cpp#L59-L62

gotcha, I thought this was referencing using 16 pins was the issue, missed that part, or comprehension of the meaning.