arduino/ArduinoCore-mbed #790: Portenta H7: Ethernet.h error with M4 core

IgnacioSJ, 2023-12-05:
Hi!
Trying to connect M4 core with a network with Ethernet, it throws me the following error:
```
C:\Users\salai\AppData\Local\Temp\arduino\sketches\CC8894BAF4C8F3E63D28F93B8DC3E5BA\libraries\Ethernet\Ethernet.cpp.o: In function `_GLOBAL__sub_I__ZN7arduino13EthernetClass5beginEPhmm':
C:\Users\salai\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\4.0.10\libraries\Ethernet\src/Ethernet.h:63: undefined reference to `EMAC::get_default_instance()'
collect2.exe: error: ld returned 1 exit status
```
If I delete Ethernet library inside mbed pack, it loads default Ethernet.h, compiles well but it doesn't work. I supposed it has to be something related with a recent update because some users have the [same error](https://community.platformio.org/t/portenta-m4-ethernet-compile-error/36717) (they say it worked for older versions). I have try to download older versions of mbed package and Ethernet.h but It doesn't work either.
Does someone know how to solve it or if M4 core allows Ethernet connections?
Thank you!

mrubioroy, 2024-04-26:
Same here. The simple code below doesn't link for M4. It does work for M7. I use Serial1 and read from UART1 pins (which works both for M4 and M7):

```
#include <PortentaEthernet.h>

void setup() {
  Serial1.begin(9600);
  while (Ethernet.begin() == 0) {
    delay(1000);
  }
  // give the Ethernet shield a second to initialize:
  delay(1000);
}

void loop() {
  Serial1.print("  DHCP assigned IP ");
  Serial1.print(Ethernet.localIP());
  Serial1.print(", DNS: ");
  Serial1.println(Ethernet.dnsServerIP());

  digitalWrite(LEDR, LOW);
  delay(100);
  digitalWrite(LEDR, HIGH);
  delay(2000);
}
```

link error:
```
/private/var/folders/gn/rm4t37bx08bgkyxy3525p1g4000200/T/arduino/sketches/887DBAF4B90ABF85F36DA98F5CF48C94/libraries/Ethernet/Ethernet.cpp.o: In function `_GLOBAL__sub_I__ZN7arduino13EthernetClass5beginEPhmm':
/Users/me/Library/Arduino15/packages/arduino/hardware/mbed_portenta/4.1.1/libraries/Ethernet/src/Ethernet.h:63: undefined reference to `EMAC::get_default_instance()'
collect2: error: ld returned 1 exit status
```

krishniot, 2024-05-07:
m4 wont support ethernet i mean you cant use ethernet related things on m4

razvanphp, 2024-09-20:
> m4 wont support ethernet i mean you cant use ethernet related things on m4

It is also (kinda) mentioned in the documentation: https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-dual-core/#m4-support

One can only assume that if Wifi does not work, Ethernet does not either.