arduino/ArduinoCore-mbed #813: A possible buffer overflow in I2C equestFrom()?

zapta, 2023-12-28:
The relevant code is in the link below. The buffer size is 256 but the the method accepts a size_t len. What happens if len is greater than 256?

https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/Wire/Wire.cpp#L94

```cpp
size_t arduino::MbedI2C::requestFrom(uint8_t address, size_t len, bool stopBit) {
	char buf[256];
	int ret = master->read(address << 1, buf, len, !stopBit);
	if (ret != 0) {
		return 0;
	}
	for (size_t i=0; i<len; i++) {
		rxBuffer.store_char(buf[i]);
	}
	return len;
}
```

multiplemonomials, 2024-01-07:
Yeah this looks like a buffer overflow to me.
And actually, looking at the code, I think that Wire could be changed to use Mbed's single-byte I2C API.  This would mean that the  we wouldn't read the bytes and buffer them, we would instead read each byte one-by-one in arduino::MbedI2C::read().

zapta, 2024-01-07:
I think that limiting the read count to 256 should be fine.  

requestFrom() is supposed to return a byte so 255 max but 256 is more intuitive when reading in chunks or pages.

https://www.arduino.cc/reference/en/language/functions/communication/wire/requestfrom/

PR #872 (2024-04-22): Wire: Fix buffer overflow in requestFrom.
Fixes #813
commit 71543378 (2024-04-22): Wire: Fix buffer overflow in requestFrom.

Fixes #813
closed 2024-04-22 by iabdalkader