arduino/ArduinoCore-mbed #817: Increase TCP connections for opta

dirkvdb, 2024-01-04:
I'm using platformio to build firmware for my opta board

I would like to increase the number of TCP connections, so I modified the `variants/OPTA/conf/mbed_app.json`
and added
```
"lwip.socket-max": 10,
"lwip.tcp-socket-max": 10,
```

No I want to compile a new libmbed.a using
`mbed-os-to-arduino-macos -a OPTA:OPTA`
but this complains that OPTA is not a valid mcu

```
Applying patches... done.
Compiling Mbed Application...argument -m/--mcu: OPTA is not a supported MCU. Supported MCUs are:
...
```

Is this not the way to modify the build for my opta board?

pennam, 2024-01-05:
Hi @dirkvdb the command is correct, and works using the linux script, probably there is a bug in the macos version.

Would you please check if this command works?

`./mbed-os-to-arduino-macos -r /path/to/my/mbed-os-fork OPTA:OPTA`

you have to download the mbed-os-fork from here: https://github.com/arduino/mbed-os/tree/extrapatches-6.17.0

dirkvdb, 2024-01-05:
Thanks!

Cloning the repo and specifying the -r options caused the build to succeed.

facchinm, 2024-01-09:
@leonardocavagnis can you take a look at why the script fails on macOS?

leonardocavagnis, 2024-01-09:
Dear @dirkvdb,
Please, could you paste the exact and complete output of the error?
I'm trying to replicate your issue.

dirkvdb, 2024-01-09:
Here it is
```
MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=0
LOCAL_REPO=
REMOTE_BRANCH=
MBED_CORE_LOCATION=/Users/dirk/.platformio/packages/framework-arduino-mbed

VARIANT=OPTA BOARD=OPTA
Checking for prerequisites... done.
Creating MbedOS Application...[mbed] Working path "/private/tmp" (directory)
[mbed] Creating new program "mbed-os-program" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#17dc3dc2e6e2817a8bd3df62f38583319f0e4fed"
 done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] OPTA now set as default target in program "mbed-os-program"
[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches... done.
Compiling Mbed Application...argument -m/--mcu: OPTA is not a supported MCU. Supported MCUs are:
ADV_WISE_1510,          ADV_WISE_1570,          ARCH_MAX, 
ARCH_PRO,               ARDUINO_NANO33BLE,      ARM_CM3DS_MPS2, 
ARM_MPS2_M0,            ARM_MPS2_M0P,           ARM_MPS2_M3, 
ARM_MPS2_M4,            ARM_MPS2_M7,            ARM_MUSCA_B1, 
ARM_MUSCA_S1,           B_L4S5I_IOT01A,         B_U585I_IOT02A, 
CY8CKIT064B0S2_4343W,   CY8CKIT_062S2_43012,    CY8CKIT_062_BLE, 
CY8CKIT_062_WIFI_BT,    CY8CPROTO_062S3_4343W,  CY8CPROTO_062_4343W, 
CYSBSYSKIT_01,          CYTFM_064B0S2_4343W,    CYW9P62S1_43012EVB_01, 
CYW9P62S1_43438EVB_01,  DISCO_F413ZH,           DISCO_F429ZI, 
DISCO_F469NI,           DISCO_F746NG,           DISCO_F769NI, 
DISCO_H747I,            DISCO_H747I_CM4,        DISCO_H747I_CM7, 
DISCO_L072CZ_LRWAN1,    DISCO_L475VG_IOT01A,    DISCO_L476VG, 
DISCO_L496AG,           DISCO_L4R9I,            DISCO_L562QE, 
DISCO_WB5MMG,           EFM32GG11_STK3701,      EFM32GG_STK3700, 
EP_AGORA,               EP_ATLAS,               EV_COG_AD3029LZ, 
EV_COG_AD4050LZ,        FF1705_L151CC,          FF_LPC546XX, 
FVP_MPS2_M0,            FVP_MPS2_M0P,           FVP_MPS2_M3, 
FVP_MPS2_M4,            FVP_MPS2_M7,            GD32_F307VG, 
GD32_F450ZI,            GR_LYCHEE,              GR_MANGO, 
HEXIWEAR,               K22F,                   K64F, 
K66F,                   K82F,                   KL25Z, 
KL43Z,                  KL46Z,                  KW41Z, 
LPC1114,                LPC1768,                LPC54114, 
LPC546XX,               MAX32620FTHR,           MAX32625MBED, 
MAX32625PICO,           MAX32630FTHR,           MAX32660EVSYS, 
MAX32670EVKIT,          MIMXRT1050_EVK,         MIMXRT1170_EVK, 
MOTE_L152RC,            MTS_DRAGONFLY_F411RE,   MTS_DRAGONFLY_F413RH, 
MTS_DRAGONFLY_L471QG,   MTS_DRAGONFLY_L496VG,   MTS_MDOT_F411RE, 
NRF52840_DK,            NRF52_DK,               NUCLEO_F070RB, 
NUCLEO_F072RB,          NUCLEO_F091RC,          NUCLEO_F103RB, 
NUCLEO_F207ZG,          NUCLEO_F303K8,          NUCLEO_F303RE, 
NUCLEO_F303ZE,          NUCLEO_F401RE,          NUCLEO_F411RE, 
NUCLEO_F412ZG,          NUCLEO_F413ZH,          NUCLEO_F429ZI, 
NUCLEO_F439ZI,          NUCLEO_F446RE,          NUCLEO_F446ZE, 
NUCLEO_F722ZE,          NUCLEO_F746ZG,          NUCLEO_F756ZG, 
NUCLEO_F767ZI,          NUCLEO_G031K8,          NUCLEO_G071RB, 
NUCLEO_G0B1RE,          NUCLEO_G431KB,          NUCLEO_G431RB, 
NUCLEO_G474RE,          NUCLEO_H723ZG,          NUCLEO_H743ZI2, 
NUCLEO_H7A3ZI_Q,        NUCLEO_L073RZ,          NUCLEO_L152RE, 
NUCLEO_L432KC,          NUCLEO_L433RC_P,        NUCLEO_L452RE_P, 
NUCLEO_L476RG,          NUCLEO_L486RG,          NUCLEO_L496ZG, 
NUCLEO_L496ZG_P,        NUCLEO_L4R5ZI,          NUCLEO_L4R5ZI_P, 
NUCLEO_L552ZE_Q,        NUCLEO_U575ZI_Q,        NUCLEO_WB15CC, 
NUCLEO_WB55RG,          NUCLEO_WL55JC,          NUMAKER_IOT_M252, 
NUMAKER_IOT_M263A,      NUMAKER_IOT_M467,       NUMAKER_IOT_M487, 
NUMAKER_PFM_M453,       NUMAKER_PFM_M487,       NUMAKER_PFM_NANO130, 
NUMAKER_PFM_NUC472,     NU_M2354,               PORTENTA_H7_M4, 
PORTENTA_H7_M7,         RHOMBIO_L476DMW1K,      RZ_A1H, 
S1SBP6A,                S5JS100,                SDP_K1, 
SDT32620B,              SDT32625B,              SDT52832B, 
SDT64B,                 SFE_ARTEMIS,            SFE_ARTEMIS_ATP, 
SFE_ARTEMIS_DK,         SFE_ARTEMIS_MODULE,     SFE_ARTEMIS_NANO, 
SFE_ARTEMIS_THING_PLUS, SFE_EDGE,               SFE_EDGE2, 
TB_SENSE_12,            TMPM46B,                TMPM4G9, 
TMPM4GR,                TMPM4KN,                TMPM4NR, 
UHURU_RAVEN,            WIO_3G,                 WIO_BG96, 
WIO_EMW3166,            XDOT_L151CC             
[mbed] ERROR: "/usr/local/opt/python@3.11/bin/python3.11" returned error.
       Code: 2
       Path: "/private/tmp/mbed-os-program"
       Command: "/usr/local/opt/python@3.11/bin/python3.11 -u /private/tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m OPTA --source . --build ./BUILD/OPTA/GCC_ARM -v"
---
```

dirkvdb, 2024-01-09:
```
cat /private/tmp/mbed-os-program/mbed-os/.git/config                                                                                Tue Jan  9 17:18:39 2024
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	ignorecase = true
	precomposeunicode = true
[remote "origin"]
	url = https://github.com/ARMmbed/mbed-os
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "latest"]
	remote = origin
	merge = refs/heads/latest
```

It works when specifying this repo with the -r option
```
cat ~/Projects/mbed-os/.git/config                                                                                                  Tue Jan  9 17:19:02 2024
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	ignorecase = true
	precomposeunicode = true
[remote "origin"]
	url = git@github.com:arduino/mbed-os.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "extrapatches-6.17.0"]
	remote = origin
	merge = refs/heads/extrapatches-6.17.0
```

leonardocavagnis, 2024-01-09:
@dirkvdb 
Try to remove `/private/tmp/mbed-os-program` folder and then run `./mbed-os-to-arduino-macos -a OPTA:OPTA`

dirkvdb, 2024-01-09:
@leonardocavagnis 
I did that to reproduce the error, otherwise it succeeded because the mbed-os git repo was still pointing to the last run with the `-r` option

leonardocavagnis, 2024-01-09:
@dirkvdb 
I'm not able to reproduce your issue. 
Please, try to clone the [ArduinoCore-mbed](https://github.com/arduino/ArduinoCore-mbed) in another location and run again the script. 
The issue could be arise because you're running the script under `/Users/<name>/.platformio/packages/framework-arduino-mbed`
Thanks

closed 2024-01-10 by leonardocavagnis