arduino/ArduinoCore-mbed #818: Support for C++17/20

Beafantles, 2024-01-05:
Hello, I’m trying to build a project for my Raspberry Pi PIco and I was wondering if I could use C++17/20 for it. However, it seems like there are some conflicts with the `abs` macro.
My `main.cpp` file is completely empty (it just includes `Arduino.h` header):

```cpp
#include "Arduino.h"
void setup() {}
void loop() {}
```

However, when I try to build the project, tons of errors are encountered, the first one being related to `abs` macro (I believe all other ones are related to that one specific error). Here’s a part of the log below:

[image]

It seems `Arduino.h` header redefines `abs` macro and it conflicts with the stdlib (for C++17/20).

AgainPsychoX, 2024-03-12:
Still not working :C

nicolube, 2024-12-19:
Hello, I too would like co compile it with C++17/20, is there any support avalible or planned?