arduino/ArduinoCore-mbed #834: LVGL 9 release

dmazzella, 2024-01-30:
The LVGL library has been updated to version 9 the changes from version 8 are in the [changelog](https://docs.lvgl.io/master/CHANGELOG.html) and Arduino IDE 2 warns of the new version release but Arduino_H7_Video library does not yet support the new display registration mode at the moment, what is your timeframe for the update?

Thanks

facchinm, 2024-02-01:
Hi @dmazzella ,
thanks for the heads up! I tried a quick 'n dirty porting, basic functionality is in place (see [here](https://github.com/facchinm/ArduinoCore-mbed/pull/new/lvgl9) and [here](https://github.com/arduino-libraries/Arduino_GigaDisplayTouch/pull/new/lvgl9) ) but there's still work to be done due to the deprecation of `sw_rotation` flag (so Giga display shield only works in portrait mode).

@kisvegabor could you provide some insights on how the new `lv_draw_sw_rotate` API is supposed to work? The original `draw_buf_rotate` implementation is quite more sensitive on RAM (and transparent).

@leonardocavagnis  since the plan is to release the core next week I'd proceed with adding an `#error` if `LVGL_VERSION_MAJOR` is `9`, so at least there's a clean way to avoid the issues flood.

kisvegabor, 2024-02-01:
Hi @facchinm,

> @kisvegabor could you provide some insights on how the new lv_draw_sw_rotate API is supposed to work? The original draw_buf_rotate implementation is quite more sensitive on RAM (and transparent).

It really required 2 separate source and destination buffers now.  Note that the "original draw_buf_rotate implementation" from LVGL v8 was dynamically allocating a smaller temporary buffer for rotation. 

The current implementation is 
- faster and cleaner
- more explicit (there is no hidden allocation)
- more flexible as you can provide any kind of destination buffer

+ you can rotate the source buffer line by line, so you need only one line for the destination buffer.

closed 2024-03-18 by leonardocavagnis

leonardocavagnis, 2024-03-18:
Dear @dmazzella,
LVGL v9 fix has been merged. It will be available in the next Core release.
Remember to update also the [Arduino_GigaDisplayTouch](https://github.com/arduino-libraries/Arduino_GigaDisplayTouch) library to version [1.0.1](https://github.com/arduino-libraries/Arduino_GigaDisplayTouch/releases/tag/1.0.1).

PR #847 (2024-03-18): Initial: support LVGL9
Fixes #834

The final patch should support both lvgl8 and lvgl9