arduino/ArduinoCore-mbed #835: Query regarding recent RPC fixes

schnoberts1, 2024-01-31:
See https://github.com/arduino/ArduinoCore-mbed/commit/a096bd6274e4a959e76e126a45332ca5d83bd605

I spent an hour today reading the 4.0.10 code to diagnose an issue with lost messages in the RPC library and quickly realised that there were some major concurrency issues. I was really pleased to see that Dec 29 these look fixed. 

I have a couple of questions: 
1. When do you think this commit will make it into a release and ...
2. ... the bug impacting me in this library was pretty basic and some easy to replicate, is there a QA issue for the ArduinoCore-mbed RPC library or is this commit a sign of an uptick in quality in the RPC library?

I'm trying to work out whether I wait for the RPC library change to be released, pull the diff between HEAD (for RPC) and 4.0.10 onto my Arduino deployment or write a simple layer on top of openamp to get out of the RPC dependency.

Any thoughts appreciated and thanks to the people working in the RPC library and fixed all this stuff. Very much appreciated.

facchinm, 2024-02-01:
Hi @schnoberts1 ,
the plan is to release early next week after merging https://github.com/arduino/ArduinoCore-mbed/pull/768 and tackling (or at leas triaging) https://github.com/arduino/ArduinoCore-mbed/issues/834 .
Thanks for your support!

schnoberts1, 2024-02-01:
Thank you for the update. Much appreciated.

closed 2024-02-01 by schnoberts1