arduino/ArduinoCore-mbed #837: Camera:  methods getResolutionWidth and getResolutionHeight are not implemented

KurtE, 2024-02-05:
In the Camera library these two methods are defined in the header file:
```
        /**
         * @brief Get the width of the current camera resolution.
         * This can for example be used to calculate the zoom window position and size.
         * In the following example, the camera is zoomed to the top right side of the image:
         * @code
         * // Calculate the zoom window position
         * uint32_t max_zoom_x = camera.getResolutionWidth() - 320;
         * // Zoom to the calculated position and size
         * camera.zoomTo(CAMERA_R320x240, max_zoom_x, 0);
         * @endcode
         * @return uint32_t The width of the camera resolution.
         */
        uint32_t getResolutionWidth();

        /**
         * @brief Get the height of the current camera resolution.
         * This can for example be used to calculate the zoom window position and size.
         * In the following example, the camera is zoomed to the bottom left side of the image:
         * @code
         * // Calculate the zoom window position
         * uint32_t max_zoom_y = camera.getResolutionHeight() - 240;
         * // Zoom to the calculated position and size
         * camera.zoomTo(CAMERA_R320x240, 0, max_zoom_y);
         * @endcode
         * @return uint32_t The height of the camera resolution.
         */
        uint32_t getResolutionHeight();

```

But they are not implemented.
They are used in one of the examples: CameraCaptureZoomPan.ino.
I tried compiling it for the GIGA, but it failed as:
```

#ifndef ARDUINO_NICLA_VISION
#error "This sketch only works on Nicla Vision."
#endif
```
And I don't have that board installed.

I was trying to use it in my own sketch, and it failed in the link phase.

iabdalkader, 2024-04-18:
> But they are not implemented.

They are implemented: https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/Camera/src/camera.cpp#L613

Maybe this issue is outdated ?

iabdalkader, 2024-04-22:
Closing this because those functions seem to be implemented, and the example builds for GIGA now. Please re-open if I missed something.

closed 2024-04-22 by iabdalkader