arduino/ArduinoCore-mbed #838: camera: FrameBuffer::FrameBuffer(int32_t x, int32_t y, int32_t bpp) might trash memory

KurtE, 2024-02-05:
In camera.cpp you have:
```
FrameBuffer::FrameBuffer(int32_t x, int32_t y, int32_t bpp) : 
    _fb_size(x*y*bpp),
    _isAllocated(true)
{
    uint8_t *buffer = (uint8_t *)malloc(x*y*bpp);
    _fb = (uint8_t *)ALIGN_PTR((uintptr_t)buffer, 32);
}

```
So if I pass in Framebuffer(320, 240, 2);
And malloc does not allocate on a 32 byte boundary, you will then increment up to the next 32 byte boundary and continue to try to 
write into an area that was not allocated to you...

If you use the default constructor without parameters, it tries to handle this properly:
```
int Camera::grabFrame(FrameBuffer &fb, uint32_t timeout)
{
    if (this->sensor == NULL
            || this->pixformat == -1
            || this->resolution == -1) {
        return -1;
    }

    uint32_t framesize = frameSize();

    if (fb.isAllocated()) {
        //A buffer has already been allocated
        //Check buffer size
        if (fb.hasFixedSize()) {
            uint32_t fbSize = fb.getBufferSize();
            if (_debug) {
                _debug->print("fbSize: ");
                _debug->println(fbSize);
            }
            if (fbSize < framesize) {
                if (_debug) {
                    _debug->println("The allocated buffer is too small!");
                }
                return -1;
            }
        }
    } else {
        uint8_t *buffer = (uint8_t *)malloc(framesize+32);
        uint8_t *alignedBuff = (uint8_t *)ALIGN_PTR((uintptr_t)buffer, 32);
        fb.setBuffer(alignedBuff);
    }
...
```

iabdalkader, 2024-04-18:
This was introduced later in https://github.com/arduino/ArduinoCore-mbed/commit/64d1a27d899de1d9110887c9241da6f6250f3216

The default constructor does handle it, however both don't handle free'ing back the memory. I guess this is fine since the framebuffer is expected to live forever.

KurtE, 2024-04-18:
Thanks,

I noticed that while I was playing with the Camera code on the GIGA.  Found there were several other issues, like missing implementations of some of the methods.  

Was not sure if best to resport those issues here or over on the Arducam github.  I ended up reporting them on the Arducam github project, and have a PR that fixes some of them.

https://github.com/ArduCAM/Arducam_dvp/pull/6

But unclear if anyone actively monitors that project.

PR #870 (2024-04-19): Camera: Fix framebuffer malloc alignment.
- Allocate extra memory for cache-line alignment.
- Fixes #838.
commit eb6f97ee (2024-04-19): Camera: Fix framebuffer malloc alignment.

- Allocate extra memory for cache-line alignment.
- Fixes #838.
closed 2024-04-19 by pennam