arduino/ArduinoCore-mbed #848: WiFi - Ethernet issue

Goliath86, 2024-03-04:
I'm creating a firmware for a OPTA WiFi. I need to connect the OPTA to a WiFi network and connect its Ethernet port to a PLC. I've noticed that if I initialize the Ethernet module first, then the WiFi everything works as expected but if I initialize the WiFi first, then the Ethernet module, the OPTA can't connect to the network. Here's an example:

```
#include <WiFi.h>

#include <SPI.h>
#include <PortentaEthernet.h>
#include <Ethernet.h>

#include "opta_info.h"

OptaBoardInfo* info;
OptaBoardInfo* boardInfo();

uint8_t *macAddress = nullptr;
uint8_t *wifiMacAddress = nullptr;

EthernetClient ethClient;
WiFiClient wifiClient;
IPAddress localIp(192, 168, 164, 40);

char ssid[] = "Tomare";
char pwd[] = "12345678";

char server[] = "google.com";

void getOptaSecureInfo() {
  info = boardInfo();
  if (info->magic == 0xB5) {
    macAddress = info->mac_address;
    String _mac = String(macAddress[0], HEX) + ":" + String(macAddress[1], HEX) + ":" + String(macAddress[2], HEX) + ":" + String(macAddress[3], HEX) + ":" + String(macAddress[4], HEX) + ":" + String(macAddress[5], HEX);
    Serial.println("Ethernet MAC address: " + _mac);
    
    if (info->_board_functionalities.wifi == 1) {
      wifiMacAddress = info->mac_address_2;
      String wifiMac = String(wifiMacAddress[0], HEX) + ":" + String(wifiMacAddress[1], HEX) + ":" + String(wifiMacAddress[2], HEX) + ":" + String(wifiMacAddress[3], HEX) + ":" + String(wifiMacAddress[4], HEX) + ":" + String(wifiMacAddress[5], HEX);
      Serial.println("WiFi MAC address: " + wifiMac);
    }
  } else {
    Serial.println("[ERR] No secure information available");
  }
}

void wifiConnect() {
  while (WiFi.status() != WL_CONNECTED) {
    digitalWrite(LED_D2, !digitalRead(LED_D2));
    WiFi.begin(ssid, pwd);
    delay(1000);
  }

  Serial.println(WiFi.localIP().toString());
  
  Serial.println("WI-FI Connected!");
  digitalWrite(LED_D2, HIGH);
}

void ethernetConnect() {
  while (!Ethernet.begin(macAddress, localIp)) {
    Serial.println("In attesa di connessione alla rete Ethernet");
    digitalWrite(LED_D1, !digitalRead(LED_D1));
    delay(1000);
  }

  Serial.println("Ethernet connected");
  digitalWrite(LED_D1, HIGH);
}

void wifiPing() {
  Serial.println("WiFi connecting...");

  if (WiFi.status() == WL_CONNECTED) {
    if (wifiClient.connect(server, 80)) {
      Serial.println("WiFi connection OK!");
      wifiClient.println("GET /search?q=arduino HTTP/1.0");
      wifiClient.println();
    } else {
      Serial.println("[ERR] WiFi connection failed!");
    }
  } else {
    Serial.println("[ERR] WiFi disconnected!");
  }
}

void ethernetPing() {
  Serial.println("Ethernet connecting...");

  if (ethClient.connect(server, 80)) {
    Serial.println("Ethernet connection OK!");
    ethClient.println("GET /search?q=arduino HTTP/1.0");
    ethClient.println();
  } else {
    Serial.println("[ERR] Ethernet connection failed!");
  }
}

void setup() {
  Serial.begin(115200);

  while(!Serial) {};

  pinMode(LED_D0, OUTPUT);
  pinMode(LED_D1, OUTPUT);
  pinMode(LED_D2, OUTPUT);

  digitalWrite(LED_D0, HIGH);

  getOptaSecureInfo();

  ethernetConnect();

  wifiConnect();

  // ethernetConnect();
  
  wifiPing();
}

void loop() {
  
}
```

The sketch present the case that does not work. If you comment out the first _ethernetConnect()_ and decomment the second one everyhing works as expected.

manchoz, 2024-03-05:
Hi @Goliath86,
I recommend switching to [Arduino_ConnectionHandler](https://github.com/arduino-libraries/Arduino_ConnectionHandler) to manage the Ethernet and WiFi connections.

Also, the Ethernet library for ArduinoCore-mbed boards doesn't actually need the MAC address (see https://github.com/arduino/ArduinoCore-mbed/blob/d53db7a37b2e61be4d5eeb86d1e9682842c57366/libraries/Ethernet/src/Ethernet.h#L79), so you can remove all the corresponding code.

Could you try using the ConnectionHandler in your sketch?

manchoz, 2024-03-07:
You welcome!