arduino/ArduinoCore-mbed #850: RPC.send() doesn't handle messages that spill a buffer

schnoberts1, 2024-03-11:
When you call `RPC.send()` the following write method is called:

`size_t RPCClass::write(const uint8_t *buf, size_t len, bool raw) {
    mutex.lock();
    OPENAMP_send(&endpoints[raw ? ENDPOINT_ID_RAW : ENDPOINT_ID_RPC], buf, len);
    mutex.unlock();
    return len;
}`

This is a one shot call and it returns the length of the buffer, i.e. the length it aspired to send. It never returns an error. 

OPENAMP_send == rpmsg_send

rpmsg_send calls `rpmsg_send_offchannel_raw(ept, ept->addr, ept->dest_addr, data,
					 len, true)`
					 
which (via indirection through a function table) calls: `rpmsg_virtio_send_offchannel_raw`.The pertinent point here is the following code. First, a TX buffer is acquired and the available space in that buffer (which is the buffer size less a header size) is written to buff_len. In the case the payload is larger than the buffer then system writes the maximum it can, buff_len bytes and buff_len is returned.  

```
	/* Get the payload buffer. */
	buffer = rpmsg_virtio_get_tx_payload_buffer(rdev, &buff_len, wait);
	if (!buffer)
		return RPMSG_ERR_NO_BUFF;

	/* Copy data to rpmsg buffer. */
	if (len > (int)buff_len)
		len = buff_len;
	io = rvdev->shbuf_io;
	status = metal_io_block_write(io, metal_io_virt_to_offset(io, buffer),
				      data, len);
	RPMSG_ASSERT(status == len, "failed to write buffer\r\n");

	return rpmsg_virtio_send_offchannel_nocopy(rdev, src, dst, buffer, len);
```

Generally in these situations a developer returns the bytes written so the caller can check if all bytes were written and if not, write what was remaining in another call. In the RPC library this does not seem to happen and instead the RPC.send() (and RPC.call()) assume all went well. RPCClass::write needs to iterate over its buffer if it's too long. 

RPCClass.send() also needs to return error codes.

schnoberts1, 2024-03-12:
The issue with splitting messages for Call and Send is it’s really a workaround for a design defect in the current API. It’s a leaky abstraction. Of course I can chop my RPC calls into N calls and static_assert sizes, but it’s suboptimal and forces every user to do the same. The same issue holds for RPC.println. 

The API also doesn’t return error codes. Why is that? I think the recent dev activity to make the RPC library work is awesome but the API itself and the implied pre and post conditions are not what they should be in a core library like this. 

> On 12 Mar 2024, at 13:18, Ibrahim Abdelkader ***@***.***> wrote:
> 
> the library doesn't split messages, or check if they fit the buffer or not, but you can easily split messages in the sketch before sending them.

schnoberts1, 2024-03-12:
What is ~32 below? Actually 32 or a varying number?

iabdalkader, 2024-03-13:
>  I think the recent dev activity to make the RPC library work is awesome but the API itself and the implied pre and post conditions are not what they should be in a core library like this.

I know there's room for improvement, please keep in mind that this is the first release of this library that actually approaches working conditions. As I've mentioned before, you're welcome to send in fixes, and if we can allocate time to work on this we will do so.

> What is ~32 below? Actually 32 or a varying number?

The actual payload is always slightly less than the RPMSG buffer size (currently 512 bytes) because some of the buffer space is used for rpmsg headers. The headers take about 32 bytes, maybe more like 40 to be on the safe side. The point is once you find that number, it's very unlikely that it will change in the future because it needs to match the other side, and even if did change the buffer will be bigger not smaller, so your app will still work.

schnoberts1, 2024-03-13:
Is there a set of automated tests I can run to verify any changes I make don’t break the library?

iabdalkader, 2024-03-13:
> Is there a set of automated tests I can run to verify any changes I make don’t break the library?

The MD5 checksum example is a good test, it should verify that everything still works if it runs for a while with no errors. It can also be used to test split buffers, by increasing the block size. If you want to, feel free to add more test cases.

closed 2024-04-18 by schnoberts1

schnoberts1, 2024-04-18:
Hi, After some investigation I closed this issue because I think the current RPC library is very far away from being useful for my use-case (from a robustness perspective) and it has turned out faster to write what I need from scratch.