arduino/ArduinoCore-mbed #858: GIGAR1: conflict between Access Point and common WiFi connection

giorgiogallina, 2024-04-04:
Hello,

I am facing some issues while trying, in a larger project, to accomplish the following:
  1. Try to connect to the internet via WiFi
  2. If connection fails, open Access Point (AP) and let user insert WiFi credentials (through captive portal)
  3. Go back to point 1

Every time I close a connection, I call `WiFi.disconnect()` and `WiFi.end()`; Moreover, when AP operations are completed, I call `WiFiClient::stop()`. My `WiFiClient` object is global and obtained through `WiFiServer::available()`.

After closing AP connection, I could connect to the WiFi but any connection (e.g., to www.google.com) failed by a time-out. I kind of solved the problem through the following code I invoke every time I switch between point 1 and 2 of the algorithm.
`WiFi = * ((WiFiClass*) (new WiFiClass(WiFiInterface::get_default_instance())));`
However I do not understand such a behaviour.

Now, if WiFi connection fails after AP was accomplished once, AP is opened again, but this time `WiFiServer::available()` fails, in the sense that no data are ever available. However, I controlled network traffic through Wireshark and I see that DNS management through `WiFiUDP` succeded, while an http request `GET /wpad.dat HTTP/1.1` was sent by the client, but never received by the server.

It seems to me that there are library issues then, or at least poor documentation for me to be able to understand whether I am doing something wrong

JAndrassy, 2024-04-04:
A WiFiClient instance is bind with network after first connect, so it will still try to use AP if it was first used with AP.
Why do you use one client object for both?

giorgiogallina, 2024-04-04:
I supposed so. Indeed, I tried to do:
```
delete server;
server = new WiFiServer(80);
```
However this seems to makes things even worse.
Thank you for your help anyway