arduino/ArduinoCore-mbed #867: Dual Core Processing on Portenta H7 using one Sketch: issue with core 4.0.10 and 4.1.1.

Hamza-Proline, 2024-04-15:
Hello everyone,

The example provided in this section '[Programming Both Cores With Just One Sketch](https://docs.arduino.cc/tutorials/portenta-h7/dual-core-processing/#programming-both-cores-with-just-one-sketch)' does not work on the M4 when the core version is 4.0.10 or 4.1.1. However, it works well with previous versions ≤ 4.0.8.

Sketch:
```
`int myLED;

void setup() {

  randomSeed(analogRead(0));
   
  #ifdef CORE_CM7  
    bootM4();  
    myLED = LEDB; // built-in blue LED
  #endif
#ifdef CORE_CM4  
  myLED = LEDG; // built-in greeen LED
#endif
pinMode(myLED, OUTPUT);
}

void loop() {
   digitalWrite(myLED, LOW); // turn the LED on 
   delay(200); 
   digitalWrite(myLED, HIGH); // turn the LED off 
   delay(rand() % 2000 + 1000); // wait for a random amount of time between 1 and 3 seconds.
}`
```

After uploading the sketch to the M7, the LEDB blinks perfectly.
Then, after uploading the sketch to the M4, the LEDG does not respond. However, if the classic Blink sketch is uploaded to the M4 it works. So the M4 boots properly via the `bootM4();` but it doesn't seem to respond to:

```
#ifdef CORE_CM4
myLED = LEDG;
#endif
```

facchinm, 2024-04-16:
Hi @Hamza-Proline ,
the issue is 99% due to the call of `analogRead(0)` from M4. Can you try to get rid of it (only for CORE_CM4) and check if everything starts working again? I remember we had a similar bug but can't find the link at the moment.

facchinm, 2024-04-16:
This patch https://github.com/arduino/ArduinoCore-mbed/commit/a758746471f018d71d8a8e7d405b886336bc8b39 might be the source of the regression; I'll investigate further and will let you know

Hamza-Proline, 2024-04-17:
Hi @facchinm , 

Exactly, after eliminating analogRead(0) from the sketch uploaded to the M4, things work perfectly.
Tested on Core version 4.1.1.
Many thanks!

tomvp, 2024-04-25:
Is this issue being worked on or is there a way to remove the offending patch?  The M4 core of my working program doesn't run If I try to use a library > 4.0.8

mrubioroy, 2024-04-26:
I confirm this issue.

facchinm, 2024-04-29:
@tomvp it's being worked on, no worries :slightly_smiling_face:

closed 2024-04-29 by facchinm