arduino/ArduinoCore-mbed #880: Giga Crashes if using ConnectionHandler and Scheduler.h

dansiviter, 2024-05-13:
If using the [ConnectionHandler](//github.com/arduino-libraries/Arduino_ConnectionHandler) example and changing it to the following:

```cpp
#include <Scheduler.h>

void setup() {
  ...
  Scheduler.startLoop(loop1);
}

void loop() {
  delay(100);  // permits yield(); to be called
}

void loop1() {
  conMan.check();
}
```
The Giga just crashes and get red LED flashing indicating crash. I'm raising it here as a potential issue with `Scheduler.h` as I tried doing effectively the same via `rtos::Thread` and it worked correctly:

```cpp
#include "mbed.h"
using namespace rtos;

void setup() {
  ...
  thread.start(loop1);
}

void loop() {
  delay(100);
}

void loop1() {
  while(true) {
    conMan.check();
  }
}
```
Frankly, there is very little within [`Scheduler.cpp`](//github.com/arduino/ArduinoCore-mbed/blob/main/libraries/Scheduler/src/Scheduler.cpp) so no idea what could be causing it.

facchinm, 2024-05-13:
I think the best way to tackle this would be to discourage the use of Scheduler lib (which was "ported" only to allow old sketches to run "seamlessly", but provides many more ways to shoot yourself in the foot than the normal, cooperative lib)

dansiviter, 2024-05-13:
Thanks @facchinm, that's sorted it. Unfortunately, my codebase needs to work on some MKR boards too and using `Scheduler.h` makes life much simpler for 3rd party libraries that like to use `delay(...)` far too much. Therefore, I'd suggest a multipronged approach: modify code to correct the stack size and also put a comment in to discourage usage. Happy to submit a PR if that's a suitable approach?

facchinm, 2024-05-13:
@dansiviter super, I'd love to merge such PR :wink:

commit dd5abb61 (2024-05-20): Scheduler: defaults stack to default OS value

* Fixes #880
closed 2024-05-20 by facchinm