arduino/ArduinoCore-mbed #887: Wrong bootloader information from factory, bricked Opta.

le0nh, 2024-05-22:
Hello, 

The problem is as follows:
 
--- Normal procedure
 
- To get the bootloader information, the instructions provided in the documentation of the arduino are used [1].
On Opta's from previous batches, this reports bootloader v24, and the information on Ethernet, USB, and other on-board features is correct.
- The bootloader of those opta's was then updated using the example sketch in the arduino IDE [2].
 
This went flawless and we have an Opta with bootloader v25.
 
--- Unexpected procedure
 
However, it seems that in certain batches of the Opta, if we try to get the bootloader information using [1],
the bootloader version reported is v255, and the information on Ethernet, USB, etc. is not correct.
 
If then the bootloader is updated using [2], the sketch executes fine, tells us we can downgrade the bootloader, and confirms that we can reboot the Opta.
When the opta is rebooted, it becomes completely unresponsive, it seems dead/bricked.
 
--- Conclusion

Not all Opta's come the same from the factory. If it is documented somewhere what is changed, and how it affects the procecedure to update the bootloader this can be acceptable.
However, currently there are Opta's that do not behave as explained in the documentation and if the steps from the docs are followed the devices get bricked.
 
This gives 2 major problems:
1). If people have a large batch of Opta's that come from the factory with bootloader v255 that need to be provisioned with bootloader v25, it seems there is currently no way to do so.
2). If people follow the Opta documentation steps they end up with a bricked device.
 
References:
[1]: https://docs.arduino.cc/tutorials/opta/mac-address/ [Date accessed: 22-05-2024]
[2]: https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/STM32H747_System/examples/STM32H747_manageBootloader/STM32H747_manageBootloader.ino [Date accessed: 22-05-2024]

alvarolb, 2024-05-24:
I got similar results from a recently received device:

[image]

Updating the bootloader seems to fix it

[image]

manchoz, 2024-06-19:
FYI @agdl, others have reported this, too.